slic3r/Slic3r #3238: bug report; slic3r crash.

Plutonium5793, 2016-02-15:
I originally reported this as a fedora bug report; they asked that I take it upstream.   My report:

This is an stl file that slic3r is having trouble with. It is an sd card holder

Description of problem:
slic3r loads a known good stl file, announces that it is too large and was automatically scaled down to fit your print bed, on saying 'ok' slic3r closes completely.
The file is 'known good' as it loads into the slic3r in Fedora 22 with no problem (not too large)

I ran from terminal with these results:

[jduchek@localhost ~]$ slic3r
Warning: File size doesn't match number of facets in the header
Caught C++ exception of unknown type at /usr/share/perl5/vendor_perl/Slic3r/GUI/Plater.pm line 1791.

Version-Release number of selected component (if applicable):
1.2.9-3.fc23

How reproducible:
always on the files it decides are too large.  Some files still load properly.

Steps to Reproduce:
1.Run slic3r
2.add affected file (stl)
3.say ok, when it tells you it has fixed the file.

Actual results:  slic3r closes so you have to start over

Expected results: display the loaded file.

Additional info: I have attached an affected stl file

Their reply:

 Miro Hrončok 2016-02-15 09:42:49 EST

I can reproduce this.

Netfabb basic says this is an invalid STL file.

ADMeshGUI silently loads the file but displays nothing.

ADMesh for command line produces the following output:

    ================= Results produced by ADMesh version 0.98.2 ================
    Input file         : sd-organizer-style.stl
    File type          : Binary STL file
    Header             : ��������������������������������������������������������������������������������
    ============== Size ==============
    Min X = -0.000000, Max X =  31139554425569280.000000
    Min Y = -4.787809, Max Y =  216341.203125
    Min Z = -0.000000, Max Z =  847403352064.000000
    ========= Facet Status ========== Original ============ Final ====
    Number of facets                 :   608                 664
    Facets with 1 disconnected edge  :    67                   0
    Facets with 2 disconnected edges :    36                   0
    Facets with 3 disconnected edges :    12                   0
    Total disconnected facets        :   115                   0
    === Processing Statistics ===     ===== Other Statistics =====
    Number of parts       :    27        Volume   :  2236782408995176448.000000
    Degenerate facets     :    19
    Edges fixed           :     0
    Facets removed        :    31
    Facets added          :    87
    Facets reversed       :   761
    Backwards edges       :     0

hexdump of a normal binary STL file looks like:

    0000000 4f43 4f4c 3d52 0000 0000 0000 0000 0000
    0000010 0000 0000 0000 0000 0000 0000 0000 0000
    *
    0000050 f06c 0000 0000 0000 0000 0000 0000 bf80
    ...

First 80 bytes are header (null terminated). Next 4 bytes are number of facets.

hexdump of your file looks like:

    0000000 ffff ffff ffff ffff ffff ffff ffff ffff
    *
    0000400 454d 4853 4d2d 5345 2d48 454d 4853 4d2d
    ...

As you can see a lot of first bytes are replaced with 0xFF. If your header is not supposed to be nonprintable garbage and number of your facets isn't max of UINT32, which the file size doesn't indicate, something is wrong with your file.

Slic3r should not crash about it. I've tested with binary from slic3r.org and also got crash (although the output in terminal is slightly different). Could you please go and report this upstream, if you think Slic3r should display a decent error message?

I have quite a few stl files that are crashing.  These are stl files I have dled from thingaverse for the most part. I have zipped the file I sent the Fedora site.
[sd-organizer-style.stl.zip](https://github.com/alexrj/Slic3r/files/130977/sd-organizer-style.stl.zip)

lordofhyphens, 2016-03-13:
From my (Debian) build of -dev:
perl: ../../src/xcb_io.c:274: poll_for_event: Assertion `!xcb_xlib_threads_sequence_lost' failed.

lordofhyphens, 2017-03-26:
This STL is complete and utter garbage. Garbage in, garbage out. 

This is in admesh. What we can do as a workaround is set the error state if we hit this specific bug (at least if the number of facets is larger than the file size) so that Slic3r doesn't try to do anything more with the file.

lordofhyphens, 2017-03-26:
I've told admesh to stop trying to hard if the file is complete garbage.

PR #3809 (2017-03-26): Abort loading if the file size is larger than the header facets
Fixes #3238
closed 2017-03-26 by lordofhyphens