slic3r/Slic3r #3239: idea: change honeycomb infill starting angles

VanessaE, 2016-02-16:
[image]

On the left is what slic3r does now on any given infill layer, at least with regular honeycomb - i.e. one of the blue paths first, then it turns around and runs through the adjacent green path, and repeat.  The back-and-forth nature of that pattern induces oscillations in the mechanics of the printer, of course.

Why not do it like the section on the right instead?  Run through the cyan paths in alternating directions, then when you're out of such paths, fill in the magenta ones.  Same number of paths over any given area of infill, but surely with much less oscillation, which would enable the user to increase their infill speed without any hardware changes.

lordofhyphens, 2016-02-16:
I'm pretty sure I'm not colorblind, but that picture looks monochrome to me.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

a4jp-com, 2016-02-16:
That might leave small air pockets after the head moves over a cooled line.

alexrj, 2016-03-13:
Interesting, I'll try it... thank you!

VanessaE, 2017-04-18:
Guess this isn't wanted or needed.

lordofhyphens, 2017-04-18:
More like nobody has time to get to it.

NateTG, 2017-04-28:
I'm not really familiar with the nuances of infill.   It seems like using cubic or triangular infill would give you better vibration reduction - is there a particular reason not to use one of those?

VanessaE, 2017-04-28:
I've not used cubic or triangular infill yet, but honeycomb is stronger than rectilinear, at least.

NateTG, 2017-04-28:
Are you printing parts where strength is an issue or doing some kind of testing?  People love to say that "hexagons are the strongest" but when I looked around for actual testing it was hard to find actual testing of 3d printed material.

https://engineerdog.com/2015/09/02/mechanical-testing-3d-printed-parts-results-and-recommendations/

The cubic infill is very similar to what you're describing with the 'diagonal' lines, but it doesn't wiggle them to make hexagons.

VanessaE, 2017-04-29:
In my particular case, it was an accidental break of a part with an upside-down "T" shape, where I had leaned against it by accident and snapped the vertical part off from the horizontal part.

When I redid it with honeycomb infill, it was more resistant to bend-and-break force than rectilinear was.

I assume this is because each pass of honeycomb infill has more filament in contact and fused with the previous layer than is the case with rectilinear.  Slic3r's "grid" infill would have probably been the strongest in that particular case.