slic3r/Slic3r #3240: Missing infill in part of a layer

forkineye, 2016-02-16:
Slic3r v1.2.9 (release) on Windows 7
Model - Bearing Blocks 01 and 02 from http://www.thingiverse.com/thing:1172838 - http://www.thingiverse.com/download:1836262
Config attached, but tried many different infill options with no resolution.

Issue:
On certain layers, no infill is being generated in part of the layer.  For the referenced bearing block model above, this is occuring on the right side, between the threaded hole and outer edge, and can first be noticed from layer 4.6 to 7.6.  It occurs again for the hole further up in the model on the same side.
[image]

lordofhyphens, 2016-03-13:
Setting perimeters and external perimeter widths to 0.45 seems to make those areas solid.

Part of the problem is that the non-external perimeters default to 0.67mm in width, which doesn't fit the thread you are trying to print. So Slic3r is doing the best it can with the restrictions you put on it (and it's probably never going to be smart enough to reduce the perimeter thickness).

Probably not actually a bug that can be fixed.

forkineye, 2020-11-20:
Closing out old issues

closed 2020-11-20 by forkineye