slic3r/Slic3r #3242: layer height always 0.3mm in g-code!

Hecot, 2016-02-16:
I'm using version 1.2.9 - when I enter different layer height it only works fine for the first layer but all the other layers will always contain 0.3mm z-movement in the g-code:

[image]

resulting g-code:

```
G1 Z0.400 F2700.000; first layer is fine!
...
G1 Z0.700 F2700.000; second should be Z0.600 !!!
...
G1 Z1.000 F2700.000; all wrong!
...
G1 Z1.300 F2700.000; all wrong!
...
```

I've tried different settings for layer hight - always 0.3mm in the g-code no matter what I enter! But different heights for the first layer work fine!!!

xoan, 2016-02-16:
Please, post your entire config by exporting from File -> Export config... Also the stl and the gcode generated file could helps a lot.

Hecot, 2016-02-18:
I've made a very simple project to have a better overview inside the g-code, please find all the files enclosed! With this simple model I found that it's the **raft layers** that are always 0.3 mm! IMO it would be good if the raft layers are of the same height as the model layers or at least be adjustable on it's own instead of an fixed 0.3 mm height!?
[10x50.zip](https://github.com/alexrj/Slic3r/files/136808/10x50.zip)

lordofhyphens, 2016-03-13:
It's not fixed to 0.3mm. It's 75% of your declared nozzle width, which is the given rule-of-thumb for maximum layer thickness.

There is no reason that I am aware of to have the raft to use model layer height, especially as the raft is designed to be discarded. Unless you've got a compelling use case, "not a bug". At best it's a low-priority feature request.

Hecot, 2016-03-14:
OK - thanks for the Feedback! I think having the same layer height on raft and part will save a lot of z-height movements ... !?

lordofhyphens, 2016-03-14:
Why? The part is always printed on top of the raft, so no matter what it'd
be the same. In fact, most people use layer heights < 75% layer height so
it'd create more individual Z moments overall.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Hecot, 2016-03-15:
Yeah, you are right - I mixed it up with supports!

closed 2016-06-20 by lordofhyphens