slic3r/Slic3r #3243: Regular print speeds overrides "slow down" cooling settings

VanessaE, 2016-02-16:
If a layer feature such as a small perimeter (the outer surface of a 5 mm diameter cylinder, for example) is bring printed, and it occurs at a point in the print where layer cooling should also be enabled, the layer fan speeds up as configured, but the print speed described in `Print Settings -> Speed` is used regardless of what speed the values in `Filament Settings -> Cooling -> Slow down if[...]` would work out to.

For example, if I have a small perimeters value of 10 mm/sec, and auto-cooling is enabled with a minimum of 1 mm/sec, any small perimeters that occur where slow-down-cooling should be active are printed at 10 mm/sec, even if Slic3r would have calculated a slower rate.

If you examine the G-code for the above example, you will notice the feed rate never drops below 600 mm/min (hence 10/sec).

Example model file: http://www.thingiverse.com/download:1121088 -- get the "lid.stl" file.
Config bundle, not that it's really needed: [Prusa i3 - Slic3r 1.3.0 config bundle.ini.zip](https://github.com/alexrj/Slic3r/files/133338/Prusa.i3.-.Slic3r.1.3.0.config.bundle.ini.zip)

Print the model by itself, using your settings or mine, but turn your layer fan speed down, as you're likely to have a better fan setup than me, which may invalidate the test.  You will notice that the little "nub" on the top of the part, and some of the surrounding slopes, will basically melt down as they're being printed.

An effective, if hacky, workaround is to add a couple of "sacrificial" objects such as tall, narrow boxes or cylinders, way off to the side, and far apart, to force the printer to slow down as desired.  This wastes filament, but adding deliberate "dwell" commands per layer would lead to ooze.

VanessaE, 2016-02-24:
duplicate of #3134

closed 2016-02-24 by VanessaE

lordofhyphens, 2016-04-10:
@VanessaE https://github.com/lordofhyphens/Slic3r/tree/debian-pr_3303 has a fix in it supplied by @hyperair 
Please give it a shot and see if it helps you.

lordofhyphens, 2016-08-24:
@KlausHans She closed it because it's a duplicate of #3134.