slic3r/Slic3r #3245: Handling inconsistent STL face normals

smurfix, 2016-02-19:
Blender's STL export script can write multiple objects into one file, but it doesn't normalize the objects consistently. I have one which, when exporting three correctly-oriented objects, consistently inverts two of these.

The problem is that, while Slic3r does determine that the object consists of three shells, it is only able to auto-invert either none or all of them. This results in an object that's only printed partially.

alexrj, 2016-03-03:
Duplicate of #1502?

smurfix, 2016-03-03:
On 03.03.2016 11:57, Alessandro Ranellucci wrote:
>
> Duplicate of #1502 <https://github.com/alexrj/Slic3r/issues/1502>?
>
>
Looks like essentially the same problem: slic3r should basically ignore
the normals and calculate the shells' orientation itself, as there's too
many broken generators out there.

-- 
-- Matthias Urlichs

lordofhyphens, 2016-03-13:
Suggest for the time being to not do multiple exports from Blender and instead use the plater if you want to make multipart STLs.

alexrj, 2016-03-18:
@smurfix, if you're reporting failure to fix nested shells we can close this one. Otherwise it's a different issue and I'd ask you to supply a test STL file.

smurfix, 2016-03-18:
On 18.03.2016 18:44, Alessandro Ranellucci wrote:
>
> @smurfix <https://github.com/smurfix>, if you're reporting failure to
> fix nested shells we can close this one. Otherwise it's a different
> issue and I'd ask you to supply a test STL file.
>
No, it's not nested. The test is easy: place two simple shells besides
each other, one oriented inside-out.

-- 
-- Matthias Urlichs

alexrj, 2016-12-22:
Note to self: in order to fix this, the `get_volume()` function in admesh should calculate volume for each shell separately. In order to fix #1502 as well, it should check whether they're nested (trickier).