slic3r/Slic3r #3246: Re-arranging objects doesn't recalculate

smurfix, 2016-02-19:
I have two large irregular objects which I want to move (partly) on top of each other so that I can save on support material. Unfortunately Slicr3r doesn't care and will happily print support material from A and the actual object B in the same place.

lordofhyphens, 2016-02-19:
Yeah, the support material is considered per model.

lordofhyphens, 2016-03-13:
I don't really know what we can do about this in terms of software. The auto-arrange is designed to place things far enough apart so this can't happen. If you saved the plate as one big STL and re-loaded it, your support probably would work as intended (as both STLs would be considered together).

mrvn, 2016-03-14:
This would be solved by issue #3266. Merge the objects and you get support for both of them combined.

alexrj, 2016-03-14:
Yes, @mrvn wrote the solution and @lordofhyphens the workaround.

closed 2017-03-18 by alexrj