slic3r/Slic3r #3248: Initial print: zigzag instead of circumference

smurfix, 2016-02-19:
My problem: my printer takes some time to really warm up and get filament flowing. I.e. initially there's too little, so it doesn't adhere to the bed. As the head travels around the mdoel, the plastic is dragged along and ends up fouling the model's bottom.

Proposed solution: a switch that tells Slic3r to zig-zag at the edge of the printable area instead of going round the model. That way, any loose plastic will stick to the just-laid track and get dragged off.

markwal, 2016-02-20:
You can probably already write your own custom start gcode to do this.  Many people do have a custom routine to purge and wipe before the beginning and a zig zag might be a relatively good way to do it.  zig zag and extrude a line down the right edge and then execute a move down the middle of the zigzag perhaps.

vlast3k, 2016-02-20:
Maybe you could try to use also G4 for adding some delay until your extruder has warmed up
E.g. G4 S20, will delay for 20 seconds

lordofhyphens, 2016-03-13:
This really isn't a slic3r problem, suggest using G4 instead in your start gcode. According to wiki, most firmwares support it.

http://reprap.org/wiki/G-code#G4:_Dwell

closed 2016-03-13 by lordofhyphens