slic3r/Slic3r #3249: Strange infill

TheNuubi, 2016-02-22:
Hi,
I recently built a second printer and everything is running quite fine right now, only the infill is wrong.
The Problem: on 100% linear infill he is doing the linear motion just fine but skips one line to then go back and fill the void (like overlapping circles).
The Honeycomb pattern looks more like a spider web. http://imgur.com/R2hdp8y.
The thing that wonders me is that in preview, layers and repetier the motion seems alright.
Firmware Marlin(stable), slicer 1.2.9,.
I have never seen someting like this, maybe i am missing something :)

lordofhyphens, 2016-03-13:
Please add the extra information (use Slic3r preview screenshot, NOT repetier host). Thanks!

Also, honeycomb is not intended to be used at 100% infill.

JFD-Mtl, 2016-03-25:
Looks like you are underextruding while doing the infill, probably because of motor skipping or filament slippage, lower the infill speeds.