slic3r/Slic3r #3253: Dangerous RAM increase when the default width extrusion is less than 21%

unconexo, 2016-02-25:
RAM increase when the default width extrusion is less than 24%. With 23% and 22% slic3r takes some time, but with 21% using up much ram and when it reaches the 90% usage of RAM, the PC freezes.

[image]

If you let it continue to grow, windows dies (BSOD).
[image]

The version is 1.2.9
Windows 10.

I do not know if it is a known bug and my google translator is not very good.
But congratulations on the job! 
Thanks.

lordofhyphens, 2016-02-26:
Need a copy of the rest of your configuration and a model in which this
happens, as mentioned in the contribution guidelines.

FYI the width is relative to the layer height when expressed in %. You
don't want to have it that low to start with.

unconexo, 2016-02-26:
True, I do not want such a low value, but I'm beginner and testing configurations that happen.
For someone else may happen and you go crazy looking for the cause.

[21 percentage extrusion with and Cube_stl.zip](https://github.com/alexrj/Slic3r/files/148335/21.percentage.extrusion.with.and.Cube_stl.zip)

If it really is not a bug, I just tried to report something that I thought would be useful. ^^

alexrj, 2016-03-13:
What is your layer height? Supposing it's 0.3mm, you're telling Slic3r to produce extrusions whose width is 0.06mm. Being so thin, you need a lot of toolpaths and your G-code file will be huge. So you also need a computer with a lot of RAM for processing this hard job, and this is not a bug.

However, a 0.06mm extrusion width is crazily small and not possible. Slic3r should definitely warn the user telling they're doing a very wrong thing.

closed 2016-03-13 by alexrj