slic3r/Slic3r #3254: V 1.2.9 Negative coordinates at bed shape

UlliBien, 2016-02-26:
Entering a minus sign in an empty fiels causes a crash. Entering a minus sign in front of an existing number works fine.

Negative coodinates are not diplayed. When opening the bed shape configuration dialog always positve values appear. But negative values seem to be stored correctly.

Size x: -200, y: -150
Origin x: -100, y: -75
=>
printer.ini
bed_shape = 100x75,-100x75,-100x-75,100x-75

=> Reopen the dialog:

Size x: 200, y: 150
Origin x: 100, y: 75

lordofhyphens, 2016-02-26:
Yeah, pretty sure it is because it tries to use the value immediately and
just a - is not a number.

lordofhyphens, 2016-03-13:
Also, why are you trying to use - sizes for the bed? That makes no sense whatsoever. - Origins are legal and work on my Linux build of current -dev, so does just putting a - in the field (no crash)

What OS are you using?

alexrj, 2016-03-18:
Fixed, thank you.

closed 2016-03-18 by alexrj