slic3r/Slic3r #3255: Retract/Layerchange order

ticso24, 2016-02-26:
...
G1 X87.435 Y122.918 E3.10626
G1 X87.901 Y122.452 E3.14222
At this point it does a Z move to the next layer height
G1 Z0.900 F9000.000
After layer change it retracts
G1 E0.14222 F1800.00000
G92 E0
G1 X136.661 Y64.760 F9000.000
...

My printer is an Ormerod 2, which has a rather slow Z movement and a bowden feeder.
The result is that I get excessive material on the layer change location, which I think could be reduced if the retract would happen before the Z movement.

lordofhyphens, 2016-02-26:
Uncheck layer change on z and do it yourself in the layer change gcode.

ticso24, 2016-02-29:
Thank you.
Your info helped me to find my stupid user error.
My fault was that retract on layer change was disabled and the retract happened because of some other reasons.
If I enable that feature everything happens in the absolutely right order.

closed 2016-02-29 by ticso24