slic3r/Slic3r #3256: feature request - solid first layer for support structures

dangasaur, 2016-02-28:
The support material first layer sparse zigzag pattern is nearly impossible to get down cleanly with PETG.  My PETG prints without support print absolutely flawlessly (first layer included) but anything that needs support takes a half dozen tries to get the first layer down cleanly.  What's worse, is that if the zigzag pattern isn't more or less contiguous then I don't even bother trying, it's a disaster.

Slic3r 1.2.9
Win7 x64

lordofhyphens, 2016-02-28:
Well, support and infill both are implemented in Perl. Could probably
splice some of the code from Fill.pm into Support as a test.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Martronic-SA, 2017-06-05:
Would it be possible to have a solid first layer for support ? (the same as interface layer?) so that the support can better stick to the bed?

Torsten85, 2019-01-19:
the firt zig zag layer is still a big issue for me... making the first support layer solid would solve this.

supermerill, 2019-01-19:
see https://github.com/supermerill/Slic3r/commit/389d3ee4d57d55bc65f2bbeb07ec89ed35618fcc (for the PE branch). I don't know if it's difficult to do/adapt for this one.

pileofrogs, 2019-04-26:
This is causing me all kinds of trouble too.

nzalog, 2019-05-23:
Yeah same problem, basically just needs an "bed interface layer" option where you can tweak the spacing.

phlfm, 2019-09-07:
I'm having the same issue with ABS. Tried using rafts for greater adhesion but my print is still failing. Would be great to have "solid first N layers"

msrd0, 2020-03-20:
Any updates on this issue? This is a nightmare every time I have to print with support