slic3r/Slic3r #3257: Feature request: Allow selection of material/extruder per part in multi material object

kyrreaa, 2016-02-29:
Right now (1.2.9) it appears assignment of materials is sequential.
This means you cannot select material 0 and 2 from your tripple setup.
You will get 0 and 1. This makes multi setups less usefull and having a head with ABS and one with PLA would be less practical if you cannot select the second extruder/feeder without using the first too.
Please, hope this can be done. It would really be usefull and might even help get multi extruders more mainstream.

lordofhyphens, 2016-02-29:
Probably not what was intended. I'll take a look over the source to see if it's an easy fix for me or not.

lordofhyphens, 2016-03-13:
From what I can tell, you can assign extruders to parts with the right-click menu, which seems like it would give you what you want. Additionally, you can assign an extruder for all print move types in the Print settings menu. 

Going to flag this as "not a bug" right now, but not close it. @kyrreaa could you provide some more information (as noted in the contributing guidelines), including a zipped copy of gcode output when you see this problem? Thanks!

kyrreaa, 2016-03-13:
Thanks lordofhyphens.
I tried the right-click -> settings thing and loaded all model parts in there, then assigned it to use 1,3,4 as extruders for a 3 part object on a printer with 4 extruders configured.
Indeed you are right it does create t0, t2 and t3 toolchanges!
This means the error is in Repetier Host which I normally use as interface to Slic3r.
It must be there the extruders are renumbered based on used materials, or the actual assignments are not passed on making it auto number.

All is well with the world!  Thanks!

closed 2016-03-13 by kyrreaa

lordofhyphens, 2016-03-14:
Can you close the issue then? :)

kefir-, 2016-03-14:
He did :)