slic3r/Slic3r #3259: Object settings used in parts/modifiers should also usable at z heights like layers tab

KriLL3, 2016-03-03:
There are a lot of uses for this, would allow you to print one of those temperature tests where you gradually change the temperature of a printed shape, would allow you to override the infill % for specific z heights which is needed frequently in order to get solid tops of non-flat topped figures etc etc, currently only way to do this is to slide your model up into separate sections and then re-assemble in slic3r's settings dialog, a very cumbersome workflow compared to just defining starting and ending z ranges to apply changes.

alexrj, 2017-03-16:
We now have a handy "Create modifier" button to create modifiers on the fly and achieve the result you described.

closed 2017-03-16 by alexrj

lordofhyphens, 2017-03-16:
Define a "slab" type mesh.