slic3r/Slic3r #3260: Unconfigurable top layer bridging extrusion width.

mirage335, 2016-03-03:
As shown in attached photo, a bit of wide-diameter infill is placed above holes closing inside of objects. In this case, a pyramid-shaped hole is 'torn' during printing.

Screenshot was taken with all extrusion widths set to 0.075mm. 

[image]

alexrj, 2016-03-13:
I don't understand the problem.
Bridge extrusion width can't be configured because of the way bridging phisically works.

You should really attach a STL file and exported config to this issue, so that we could reproduce and understand better.

mirage335, 2016-03-13:
The bridging layer is too thick. Not just where actual briging occurrs, but for the entire region with large lines is over-extruded, as shown. Would be nice to reduce the extent of and area covered by over-extrusion when fine detail in previous layers is likely to be damaged.

For now, I have worked around this by attempting to print the object vertically.

Slic3r configuration available at:
https://github.com/mirage335/TazStiff-Softload/tree/master/TazStiff_Config/.Slic3r

STL file available at:
https://github.com/HacDC/HacDCOptics/blob/master/_exports/STL/FiveAxis_Substages.stl

lordofhyphens, 2016-08-24:
STL disappeared, could you zip it and an exported config.ini file (load slic3r, load configuration combination you see this problem with, File->Export Config) together and drop it in the issue? This is very important for reproduction.

Not going to try every combination of print/filament/printer configuration.

mirage335, 2017-12-30:
STL file is still in the git commit history.
https://github.com/HacDC/HacDCOptics/blob/6491414cd555b975028db66619c2271260cf0ac2/_exports/STL/FiveAxis_Substages.stl
Slic3r configuration at the time was probably reflected by this commit.
https://github.com/mirage335/TazStiff-Softload/tree/21927f91e215b7e31fae9d0cd7598f20c2c0f7ea/TazStiff_Config/.Slic3r

IIRC, all combinations of print/filament/printer configuration in that configuration directory were causing the issue. However, enough time has passed that it is possible the latest version of Slic3r today has already fixed the issue.