slic3r/Slic3r #3261: 0-speed moves when support material speed is 0

hyperair, 2016-03-06:
Slic3r generates G-code with `G1 .... F0` moves when support material speed is set to 0. This is a source of confusion for users who are enabling autospeed for the first time, because they just set everything to 0 without realizing that support material speed doesn't have 0=autospeed. Same goes for support material interface speed, and bridge speed.

hyperair, 2016-03-06:
Okay, this seems to be a floating point issue. I edited `Slic3r::Print::GCode::BUILD` to print out `$min_mm3_per_mm` and got `4.44089209850061e-15`. A couple lines up there's a `grep` call filtering out 0 values:

```perl
            @mm3_per_mm = grep $_ != 0, @mm3_per_mm;
```

Equality checks don't work on floats. Changing this to `$_ < 0.00001` works.

alexrj, 2016-03-13:
Do you have a failing test case for this? I'd like to test it before applying a fix.
Support material speed is actually supposed to use autospeed, despite the tooltip doesn't mention it.

hyperair, 2016-03-14:
I do, but the STL isn't mine to provide. I haven't been able to come up with another test STL that reproduces the issue. If you look at the diff in the pull request though, the part that has been fixed is quite clearly wrong though -- there's a filter for != 0 there, which won't work with floats.

lordofhyphens, 2016-05-17:
@hyperair could you provide it privately for testing purposes (so it's not on a public forum)? I'll try to talk to you on IRC later.

PR #3262 (2016-05-19): Don't use equality comparisons for floats
This fixes an issue where F0 moves arise from 0-width (or 0 layer height?) support material segments when using autospeed.

Fixes: #3261
commit 5f8eea9a (2016-05-19): Don't use equality comparisons for floats

This fixes an issue where F0 moves arise from 0-width (or 0 layer
height?) support material segments when using autospeed.

Fixes: #3261
closed 2016-05-19 by lordofhyphens