slic3r/Slic3r #3263: Gaps between infill and perimeters at small layer hights

MartinSausF, 2016-03-07:
I wanted to print with 0.04mm layer hight and got gaps between infill and perimeters. With 0.08 layer hight the gaps are gone. Maybe slic3r calculates with big extrusion width because of the small layer hight but I set combine infill every 4 layers so the extrusion width is maybe smaller than slic3r expects at 0.04mm layer hights!?

0.04mm layer hights:
[image]

0.08mm layer hights:
[image]

I use PLA and 0.3mm nozzle.

lordofhyphens, 2016-03-07:
Not to sound like a broken record, but please read the contribution
guidelines and post the extra information requested in them.

lordofhyphens, 2016-03-11:
That also looks like pictures from RepetierHost, which has no clue what the actual widths are and so can't be trusted. What does Slic3r's own preview UI tell you?

MartinSausF, 2016-03-14:
Hi, sorry for the delay! Here are more infos and screenshots from the slic3r ui:

Slic3r Version 1.2.9
Operating System: Win7 Professional SP1 and Win10 Pro Version1511
Steps for reproduction:
- Add any solid part like cube 20x20mm
- use [config.zip](https://github.com/alexrj/Slic3r/files/171655/config.zip)
- most important settings (already done in config-file):
* nozzle diameter 0.3mm
* layer height: 0.04mm
* infill 80% Rectilinear
* combine infill every 5 layers

Screenshots:
Layer1(bottom layer):
[image]

Layer5 (one of the first infill layers):
[image]

Layer11 (second infill layer):
[image]

My assumption what happend at very small layer high and combined infill:
First Infill Layer after bottom layers gets very wide because of the very small layer height. Due to the combined infill in the next infill layers the width of the infill gets "normal" but the distance between the infill "center line" and the perimeters stays the same and generates a gap.

There are also some grafically issues (especially with the infill) at small layer heights like 0.04mm.

Sorry for my little bad english (i am from germany ;))

MartinSausF, 2016-04-25:
Any Feedback?!

lordofhyphens, 2016-04-25:
Not that surprised w/r/t combined infill giving a few issues. I haven't had
time to reproduce your issue.

Have you played around with the infill widths or the number of combined
infill layers?