slic3r/Slic3r #3264: between objects (custom) G-code

mrvn, 2016-03-08:
When printing objects sequentially slic3r sets the first layer temperature for every objects first layer. But it does not wait for the temperature to be reached. It would be nice if slic3r would move the head to somewhere between objects but keep it at least 5mm above the bed, then set and wait for the first layer temperature and last print a (mini) skirt around the next object or extrude some extra material to refill the melt chamber after the pause.

On a more generic front the Print Settings->Custom G-code page should have a box for custom G-code between objects, which would get triggered somewhere in the empty space between objects and before lowering the head.

lordofhyphens, 2016-03-17:
Unfortunately gcode to wait until something has cooled to a specific temp is a bit inconsistent. Agreed on the ability to add custom gcode between objects. I think the more predicatable point is after the completion of an object. 

I am a little concerned about UI clutter, though.

lordofhyphens, 2016-03-18:
I've got something that would work if I only knew how to add items to ConfigOptionsGroup :( I'll have to cook up a test for it.

@alexrj care to give me a pointer? I can't seem to find things (I'll upload a PR so you can see my changes as-they-are).

mrvn, 2016-03-18:
It's not about cooling (not that that couldn't happen too :), it's about heating up. I have my first layer temp higher than the rest. But I assume "M109 <first_layer_temp>" (or what the magic syntax was in the before G-code) would work in your solution too?

lordofhyphens, 2016-03-18:
@mrvn actually that's the inconsistent part across firmwares. A few only wait until it's met or exceeded temp. But hey, working on it :)

lordofhyphens, 2016-05-29:
@mrvn there's a pull request now, I don't have a proper test case for it though (I can't grok the test harness system). I am going to close this now (I added the PR to the milestone)

closed 2016-05-29 by lordofhyphens

Mikk36, 2016-05-29:
Pull request in question: #3337

alexrj, 2017-04-04:
This is merged as of bc96a1a, and available in the automated builds at http://dl.slic3r.org/dev/