slic3r/Slic3r #3266: Feature Request: merge objects, change object ordering, rename objects

mrvn, 2016-03-08:
Sorry for combining multiple feature requests but they kind of concern the same problem.

I like printing objects sequentially. But sometimes not all objects can be printed sequentially. There might not be enough space or some objects are too high.

If there are too many objects to leave sufficient space between them it would be nice if one could select multiple objects and merge them into a single object. So for example when I want to print 16 objects I could print them 4 at a time.

In other cases on of the objects is too high to print sequentially. Except if it's the last object in the list the height does not matter. Or I simply want to print the most problematic part first. But when splitting an multi-part object one doesn't get a choice in the order. It would be nice if one could change the order in which parts are printed (move them up or down in the list).

Last it would be nice to rename objects in the list, especially after splitting a multi-part object. Just so it's easier to see which object is which.

alexrj, 2016-03-13:
All good ideas (reordering objects is already tracked by one or more existing issues here).

syl405, 2017-10-13:
Now working on implementing this.