slic3r/Slic3r #3268: Slic3r using M140 instead of M190 after first layer, not waiting for bed to heat up.

GreatGrizzly, 2016-03-09:
Slic3r is using M140 S75 to wait for the bed temperature to be reached after the first layer. According to the GCode Wiki, M140 means to NOT wait for the bed to reach temperature. 

M190 should be used instead. 

This also contradicts the very comment in the GCode that was created by Slic3r: 

> M140 S75 ; wait for bed temperature to be reached

I attached the GCode file that slic3r generated.
[Filter.gcode.txt](https://github.com/alexrj/Slic3r/files/165668/Filter.gcode.txt)

lordofhyphens, 2016-03-09:
Please read the contributing guidelines and follow them, you have not
provided necessary information.

https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

GreatGrizzly, 2016-03-09:
Added more info.

lordofhyphens, 2016-03-09:
It's doing exactly what your start gcode is telling it to do. So fix that. One of the printer-specific things that needs to be configured. 

start_gcode = ;Start heating up everything\nM140 S[first_layer_bed_temperature]  ;Set bed tempurature, continue on.\nM104 S[first_layer_temperature]

GreatGrizzly, 2016-03-09:
Here is what I want to do:

1) Start heating up bed and hotend at same time. 
2)Do some things. 
3) Wait for bed and hotend to reach the same temperature as before. 
4) Print all of the first layer
5) Start heating up bed and **wait for it to complete**, then start second layer. 

**Currently the GCode is generated like this:**

M140 S25 ;Start heating up the bed to save time....
<Do Calibration Stuff>
M190 S25 ;The bed is already in the process of heating up, now the printer will wait for it to complete.
<Print First Layer>
M140 S75 ;This sets the bed to 75, but continues printing even though the bed is not hot enough yet!

**What it needs to be done:**
M140 S25 ;Start heating up the bed to save time....
<Do Calibration Stuff>
M190 S25 ;The bed is already in the process of heating up, now the printer will wait for it to complete.
<Print First Layer>
**M190** S75 ;This sets the bed to 75, and waits for it to reach target temperature before starting the second layer.

This M190 functionality is literally in the [GCode Wiki](http://reprap.org/wiki/G-code)

markwal, 2016-03-09:
@GreatGrizzly If you put the expansion tokens in your start gcode, it'll supress the automatic generation of anything.  But, if you want it to wait for S75, why don't you just put it in your start gcode?

M140 S25
M190 S25
M190 S75

Then even if Slic3r puts an M140 S75 after that it won't matter because it already waited.  But like I say, you can even surpress that if you use expansion tokens in your gcode.  Mine, for example has:

M140 S[first_layer_bed_temperature]
M104 S[first_layer_temperature_0] T0
; somestuff, then
M116 ; wait for all target temps

lordofhyphens, 2016-03-09:
Printer Settings->Start gcode if you are on simple mode.

Printer Settings->Custom Gcode->Start Gcode if on expert.

This is not a slic3r issue. You can edit the order of the start gcode to do
what you want. Your start gcode already has the correct bits in it to
suppress autogeneration of the heat up gcode.

GreatGrizzly, 2016-03-09:
> if you put the expansion tokens in your start gcode, it'll supress the automatic generation of anything. But, if you want it to wait for S75, why don't you just put it in your start gcode?

BECAUSE I DONT WANT THE BED TO BE AT 75C DURING THE FIRST LAYER, only all the layers after the first... a setting that is in the Slic3r.

> This is not a slic3r issue. You can edit the order of the start gcode to do
what you want. Your start gcode already has the correct bits in it to
suppress autogeneration of the heat up gcode.

There is no "Custom GCode after first layer only" option in slicer. The temperature change between the first layer and the second layer code is generated by Slic3r. This code is where the M140 line is. 

Did you even look at the gcode? Even the comment at the end of the M140 command is WRONG.

> M140 S75 ; wait for bed temperature to be reached

The wiki entry on M140:

> Set the temperature of the build bed to 55oC and **return control to the host immediately** (i.e. before that temperature has been reached by the bed). There is an optional R field that sets the bed standby temperature: M140 S65 R40.

lordofhyphens, 2016-03-09:
Set it in the between layer gcode section then if you want it colder on layer 1 and hotter on layer 2 (this is kind of a weird situation btw). Or write a post-process script to do the same thing.

lordofhyphens, 2016-03-09:
@GreatGrizzly As far as I know from a fresh install, Slic3r doesn't ship with any of that start gcode, so that the comments are wrong means nothing. I don't know where you got it from , but the only start gcode that ships with slicer is

G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

markwal, 2016-03-09:
Like @lordofhyphens lordofhyphens said, then you can do it in the layer change with the same technique.

By the way, I actually use the layer change to reduce my bed temperature between layer 1 and layer 2 and would prefer that it *not wait*.  Since people need different things then we need the slicer to be customized for our needs.

It turns out we already have all the flexibility we need to have it do what we want due to the custom gcode (in this case, layer change).  Which is good, because at this point, if you aren't willing to modify the source code yourself, you're out of luck.

lordofhyphens, 2016-03-09:
@markwal yeah, some firmwares wait for cooling on M190/M109 and some don't too, which adds to the *fun*.

mrvn, 2016-03-10:
the first layer is special and one might not want to do something after every layer just to get an effect after the first layer. Maybe slic3r should have the option to add custom G-code for after the first layer. G-code after raft would also be nice when one wants to change filaments there (e.g water soluable raft). More generic would be G-code after layer <X>, with the option to add any number of them.

lordofhyphens, 2016-05-29:
@mrvn just pasting in gcode after some arbitrary point is probably better left to post-processing scripts. With verbose gcode on it's pretty trivial to find & count layer changes.

lordofhyphens, 2016-07-11:
@GreatGrizzly PR #3411 will do what you want. Defaulting to true is a 1-line change to GCode.pm, line 378 (add ,1 to the function call).

SphaeroX, 2017-09-18:
i got the Problem that Slic3r uses 

> M140 S55 ; set bed temp

than my anet A8 with Maril 1.1.5 set the Hotbed to 130, when i change this to like Cura

> M190 S55

it works correctly

lordofhyphens, 2017-09-18:
@SphaeroX If this is in your startup gcode, then configure it properly. If you put 
````
M190 S[first_layer_bed_temperature]
````
 into your start gcode (Printer settings->Custom Gcode) then it won't add the M140 on its own. 
Add ````M109 S[first_layer_temperature]```` and it'll set your hotend temp too. 

I don't know why your firmware would be trying to set the bed temp to 130 with M140 call; it isn't Slic3r's fault if your firmware is doing weird things.

commit ea29a22c (2018-04-13): Adds Printer config option to force the use of set-and-wait gcode semantics (if supported by firmware).

Fixes #3268
closed 2018-04-13 by lordofhyphens