slic3r/Slic3r #3270: XS configure Fail while installing slic3r dependencies

SphereCo, 2016-03-12:
Hi,
I am trying to run slic3r from source on windows and I have been struggling for quite some time now... I have followed the guidelines "Running Slic3r from Git on Windows" and while installing the dependencies, I just can't get XS to install properly.

Slic3r is current clone from git, master branch
OS is Microsoft Windows [Version 10.0.10586]
Boost is 1_60_0 (path: C:\dev\boost_1_60_0)
StrawberryPerl is 5.22.1.3-64bit

when running perl Build.PL (in slic3r repository), I get the following:

```
C:\Slic3r-master>perl Build.PL
App::cpanminus is up to date. (1.7040)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.06)
Encode is up to date. (2.82)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.10)
ExtUtils::ParseXS is up to date. (3.30)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.62)
Getopt::Long is up to date. (2.48)
IO::Scalar is up to date. (2.111)
Math::PlanePath is up to date. (122)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.001001)
POSIX is up to date. (1.53_01)
Scalar::Util is up to date. (1.43)
Test::Harness is up to date. (3.36)
Test::More is up to date. (1.001014)
Thread::Semaphore is up to date. (2.12)
Time::HiRes is up to date. (1.9728)
Unicode::Normalize is up to date. (1.25)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
XML::SAX::ExpatXS is up to date. (1.33)
threads is up to date. (2.02)
cpanm (App::cpanminus) 1.7040 on perl 5.022001 built for MSWin32-x64-multi-thread
Work directory is C:\Users\JF/.cpanm/work/1457810568.13132
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.15
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/Slic3r-master/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4216)
Running Build.PL
Configuring C:/Slic3r-master/xs ...
Slic3r requires the Boost libraries. Please make sure they are installed.

If they are installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply their
path through the BOOST_DIR environment variable:

    BOOST_DIR=/path/to/boost perl Build.PL

N/A
! Configure failed for ./xs. See C:\Users\JF\.cpanm\work\1457810568.13132\build.log for details.
The XS/C++ code failed to compile, aborting

C:\Slic3r-master>
```

and the build log is not of much help...
```
cpanm (App::cpanminus) 1.7040 on perl 5.022001 built for MSWin32-x64-multi-thread
Work directory is C:\Users\JF/.cpanm/work/1457810568.13132
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.15
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/Slic3r-master/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4216)
Configuring C:/Slic3r-master/xs
Running Build.PL
-> N/A
-> FAIL Configure failed for ./xs. See C:\Users\JF\.cpanm\work\1457810568.13132\build.log for details.
```

I just can't figure what is going wrong with the XS configure.

lordofhyphens, 2016-03-12:
Did you set the variable so that Build.pl can find the boost libs?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

JFD-Mtl, 2016-03-12:
Tried to, in XS/Build.PL, I don't think I succeeded tho. I must admit I just don't know how to set this variable... I tried to find answers to "how to set BOOST_DIR variable" in google but without success...

I thought that putting it in C:/dev/ was enough, especially since it is supposed to be smart enough to find it.

lordofhyphens, 2016-03-12:
Have you tried with boost 1.55 instead of 1.60?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

JFD-Mtl, 2016-03-13:
I tried with 1.59 and it did buid
now I am stuck installing the gui, wx-0.9928 won't work. It's stuck at what I guess is the testing, one of the test windows called "X" keeps getting stuck and goes not responding. shutting it down makes the installation fail.

alexrj, 2016-03-13:
Type `cpanm -f Wx`, it will force the installation of the Wx module ignoring the failing test. Then you can repeat the `Build.PL --gui` command.

JFD-Mtl, 2016-03-13:
Awesome, it worked.

I guess the only issue remaining is the fact that it won't compile with boost 1.60 while it does with 1.59

also, as a site note, the wx bug seems to only be affecting some compilers but not all: https://rt.cpan.org/Public/Bug/Display.html?id=94519

thanks alex!

lordofhyphens, 2016-03-13:
Not compiling with different libs is something that is not entirely unreasonable :). We can get the compile instructions updated to indicate that it is not currently compatible with 1.60.

closed 2016-03-25 by JFD-Mtl

markwal, 2016-03-25:
boost 1.60 works for me now

JFD-Mtl, 2016-03-25:
Good to know, I'll try it

markwal, 2016-03-25:
I force installed Wx though and had to kill that stuck window. That part doesn't have anything to do with boost. The boost error was just that xs compile error caused by a incompatible bug fix in boost that Allesandro worked around.

paulorfmmb, 2017-02-06:
I'm having the same problem where I try to force install Wx but the slic3r.pl breaks when I try to open it.

lordofhyphens, 2017-02-06:
For 32 bit perl I have a "slic3r dependency" archive on bintray.com you can
use.

Or just get the auto build.

lordofhyphens, 2017-02-07:
@paulorfmmb Also, please don't comment on closed issues. It's extremely rare that someone will ever notice. 

For dependencies:
https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-perl-dependencies-5.24.0-win32.7z

perlwx is known to be a pain to compile on Win32 systems, especially if you have multiple compilers installed.