slic3r/Slic3r #3271: Warn users when they're using possibly wrong values

alexrj, 2016-03-13:
Including too thin extrusion widths (see #3253)

mrvn, 2016-03-14:
Actually I'm confused that the extrusion width is computed over the layer height. Say I have it at 100% and I print 0.3 mm layers then it's maybe reasonable. Now I select 0.05 mm layers and it is totally insane.

Imho extrusion width should be based on nozzel size because that is the main deciding factor. There is no way a 0.5 mm nozzle can extrude 0.1 mm perimeter accurately. As is % values for extrusion width make no sense to me.

alexrj, 2016-03-14:
A few years ago, the concept of width-over-height or width-over-thickness was quite popular in RepRap software and literature. Width over nozzle diameter does not make much sense. Setting absolute values or using the automatic algorithm are the recommended ways (anyway, this if off topic in this issue, better use the original one).

lordofhyphens, 2016-03-14:
Something that I think would be convenient/useful would be a min/max (in
absolute terms) line width. A setting of 0 for either would use a default
(min would be nozzle width, max would be no limit).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

unconexo, 2016-03-14:
That's my point, thanks for making that request feature. :)
Regards