slic3r/Slic3r #3274: Move Debian branch to main repo?

lordofhyphens, 2016-03-18:
Is there interest in moving my debian repo build source to a branch of the main repo, or should it remain in my fork? For users of Debian-based distros it's likely to be helpful to people that want the bleeding edge but don't want to risk breaking their own local Perl stuff by compiling from scratch.

In related news my debian branch has been rebased (and will be rebased in the future instead of merging; this should keep things cleaner).

AndKe, 2016-04-12:
yes, sounds good.

alexrj, 2017-03-18:
Are you still maintaining the Debian repo? Anyway, go for it!

AndKe, 2017-03-19:
Another thing to consider , would be to makea snap package, should make it much easier for all kinds of distros.