slic3r/Slic3r #3278: test failing on current master

wolfmanjm, 2016-03-19:
t/thin.t ....................... 1/21 
#   Failed test 'medial axis has reasonable length'
#   at t/thin.t line 106.

#   Failed test 'medial axis has reasonable length'
#   at t/thin.t line 121.

#   Failed test 'medial axis has a reasonable length'
#   at t/thin.t line 133.

#   Failed test 'medial axis of a narrow triangle is a single line'
#   at t/thin.t line 143.
#          got: '0'
#     expected: '1'
Can't call method "length" on an undefined value at t/thin.t line 144.
# Looks like you planned 21 tests but ran 15.
# Looks like you failed 4 tests of 15 run.
# Looks like your test exited with 255 just after 15.
t/thin.t ....................... Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 10/21 subtests 
t/threads.t .................... ok   
t/vibrationlimit.t ............. ok   

Test Summary Report
-------------------
t/thin.t                     (Wstat: 65280 Tests: 15 Failed: 4)
  Failed tests:  11, 13-15
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 21 tests but ran 15.
Files=33, Tests=454, 119 wallclock secs ( 0.15 usr  0.06 sys + 129.06 cusr  1.42 csys = 130.69 CPU)
Result: FAIL

a4jp-com, 2016-03-21:
Same here on Windows 10 (64-bit)

I just downloaded an hour ago but still can't compile 1.3.0-dev latest commit e4147a6

`Microsoft Windows [Version 10.0.10586]
(c) 2015 Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>cd..

C:\Windows>cd..

C:\>cd slic3r-4

C:\slic3r-4>perl Build.pl --v
App::cpanminus is up to date. (1.7040)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.06)
Encode is up to date. (2.82)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.10)
ExtUtils::ParseXS is up to date. (3.30)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.62)
Getopt::Long is up to date. (2.48)
IO::Scalar is up to date. (2.111)
Math::PlanePath is up to date. (122)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.001001)
skipping S/SH/SHAY/perl-5.22.1.tar.bz2
Scalar::Util is up to date. (1.44)
Test::Harness is up to date. (3.36)
Test::More is up to date. (1.001014)
Thread::Semaphore is up to date. (2.12)
Time::HiRes is up to date. (1.9732)
Unicode::Normalize is up to date. (1.25)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
XML::SAX::ExpatXS is up to date. (1.33)
threads is up to date. (2.02)
Cleaning up build files
Cleaning up configuration files
cpanm (App::cpanminus) 1.7039 on perl 5.022000 built for MSWin32-x86-multi-thread-64int
Work directory is C:\Users\design/.cpanm/work/1458531844.12828
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.15
Falling back to Archive::Tar 2.04
--> Working on ./xs
Entering C:/slic3r-4/xs
Checking if you have Module::Build 0.38 ... Yes (0.4212)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Running Build.PL
Configuring C:/slic3r-4/xs ... 'cc' is not recognized as an internal or external command,
operable program or batch file.
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Module::Build 0.3601 ... Yes (0.4212)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\admesh\normals.o" "src\admesh\normals.c"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\libslic3r\PrintRegion.o" "src\libslic3r\PrintRegion.cpp"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\admesh\stl_io.o" "src\admesh\stl_io.c"
gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -IC:\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\libslic3r\ExPolygon.o" "src\libslic3r\ExPolygon.cpp"
[... 162 lines cut ...]

#   Failed test 'medial axis of a narrow triangle is a single line'
#   at t\thin.t line 143.
#          got: '0'
#     expected: '1'
Can't call method "length" on an undefined value at t\thin.t line 144.
# Looks like you planned 21 tests but ran 15.
# Looks like you failed 4 tests of 15 run.
# Looks like your test exited with 255 just after 15.
t\thin.t ....................... Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 10/21 subtests
t\threads.t .................... ok
t\vibrationlimit.t ............. ok

Test Summary Report
-------------------
t\thin.t                     (Wstat: 65280 Tests: 15 Failed: 4)
  Failed tests:  11, 13-15
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 21 tests but ran 15.
Files=33, Tests=454, 130 wallclock secs ( 0.09 usr +  0.03 sys =  0.13 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

C:\slic3r-4>`

lordofhyphens, 2016-03-21:
@a4jp-com could you please upload build logs as .txt files and not paste
right into the box? It makes it really hard to read because of github
interpreting things as markdown and makes the email interface really
annoying to use.

It also makes scrolling through the issue take a lot longer than it should
on the issue page. Thanks.

a4jp-com, 2016-03-21:
Okay. Sorry about that. I'll do that from now on if I paste a huge chunk of code etc.

Any advice on fixing the files so I can compile the source code?

lordofhyphens, 2016-03-21:
Those tests look related to new changes; @alexjr could probably tell you if
you can ignore them.

alexrj, 2016-03-21:
The test fix is in the works. @a4jp-com, that's not a compile failure. You got it compiled, so you can use it. I have a strict stable policy for the `stable` branch, but I tend to be more relaxed about `master` so some tests sometimes might fail.

a4jp-com, 2016-03-21:
Very cool. I just thought it wouldn't work. I'll try the --gui bit now.

a4jp-com, 2016-03-24:
Maybe this can be closed as the software can be compiled even though it has errors.

@wolfmanjm all you need to do after `perl Build.PL --v` is `perl Build.PL --gui`

It has errors and my model was really strange but the program is at least running.

[image]

alexrj, 2016-03-26:
Test was fixed.

closed 2016-03-26 by alexrj