slic3r/Slic3r #3279: [1.3.0-dev] Perimeters laid in mid-air

lordofhyphens, 2016-03-19:
[floating_perimeters.zip](https://github.com/alexrj/Slic3r/files/181060/floating_perimeters.zip)
as of commit dfb1ec99ac2699840e0dbe40163fade9df3e966b

If you fix Infill, Perimeters, and External perimeters width to 0.6, you start seeing strange artifacts around the same point in the model.

lordofhyphens, 2016-03-19:
It's commit dfb1ec99ac2699840e0dbe40163fade9df3e966b that is causing the behavior, as reverting it fixes the behavior.

alexrj, 2016-03-26:
I'm not sure I can see the actual problem in your screenshot, but could you test as of current master? I pushed many fixes.

alexrj, 2017-03-14:
@lordofhyphens, can you confirm this is fixed?

lordofhyphens, 2017-03-16:
Unfortunately it's not.
Here's the same area at 3 different heights:

This is the last area touched by infill.
[image]

As can be seen, several perimeters (but no infill) get built up here on the intervening layers for a full mm.
[image]

The perimeter now appears between the two perimeters that had been laid up is floating in the air.
[image]

A closer look.
[image]

lordofhyphens, 2017-03-16:
I found a second shell in the provided test case, a single ring. It looks like it is the confounding agent here--the gap is being formed because Slic3r is somehow interpreting the second shell (a ring) in a weird way. If you turn on support, Slic3r tries to generate supports in the gap. 

Splitting the model eliminates the symptoms. I'm not sure where to put this on the bug hierarchy, as I don't recall putting that ring in there when I SCAD'ed up the model. 

Up to you how you want to handle it w/r/t tagging not a bug, as the model is showing as manifold.

alexrj, 2017-03-26:
Actually, the model is manifold, and Slic3r _is_ doing the right thing!

It's not just two nested shells (in which case they would be considered as a single solid): the outer shell has a hole/void, and the inner shell is inside that void. So they are actually two nested and non-overlapping *solids*. The object is modeled as if you wanted a circular tunnel inside the object, and that's what Slic3r is doing.

 I'm attaching some more screenshots showing the situation:

[image]
[image]
[image]

closed 2017-03-26 by alexrj