slic3r/Slic3r #3280: slic3r 1.2.9 ignoring print centre values when slicing through the commandline

jgrjgr, 2016-03-20:
Hi all,

there is an issue with 1.2.9 where under the commandline it produces gcode based on an assumption that the print centre is X100 Y100 which is fine for most reprap beds, however if you bed is a bit larger eg  400x200 it still outputs gcode centred around X100 Y100 when it should be centred around the print centre values which are 200x100

I've managed to reproduce this across windows: 7 , 8.1 and 10 machines on 32bit and 64bit systems
all relavant files are here: https://www.dropbox.com/sh/8flsems65epn82z/AABKjGkJF3WgkLkkr4c3W1MJa?dl=0

the commandline usage was:

slic3r-console.exe cube.stl -load config-big.ini

alexrj, 2016-03-20:
Thank you, duplicate of #2725

closed 2016-03-20 by alexrj

jgrjgr, 2016-03-20:
that issue is about the value being generated dynamically based on bed size, i'm talking about it ignoring the values placed in the config.ini