slic3r/Slic3r #3281: Gaps filled over infill

Vicious-one, 2016-03-20:
Sometimes the infill is generated in narrow areas even when gap fill is enabled.

[image]

STL and config:
[gap-double-fill.zip](https://github.com/alexrj/Slic3r/files/181357/gap-double-fill.zip)

markwal, 2016-03-20:
Which version are you running?  There was a fix for a similar bug, but if you're running a new version, it could be a regression since the most recent changes were also in the area of gap fill.

@lordofhyphens maybe it's time for a git issue template?

Vicious-one, 2016-03-20:
@markwal -- I've just built it from master. But this issue is kinda long-standing.

markwal, 2016-03-20:
Yes, but there were a bunch of recent fixes for precisely this issue, so we need to figure out whether you're seeing the same issue for a different cause, don't have the latest, or still see an unfixed issue.  Also, when giving version info, a commit number is more useful than "master" because "master" changes.

You can use:

git describe --tag

to git a descriptive version number, but report it like that:

git describe --tag says "1.2.9-221-g5ff7511"

Vicious-one, 2016-03-20:
@markwal -- Done

lordofhyphens, 2016-03-20:
@markwal got one up, probably could be better though. #3284 3284

Vicious-one, 2016-03-20:
When Gap fill is disabled, the infill also seems strange:

[image]

Vicious-one, 2016-03-27:
The original issue seems to be fixed in 1.2.9-234-g9576973
(At least with that model)
[image]

alexrj, 2016-12-21:
Yes, it was fixed! Thank you for reporting, test cases are always very useful.

closed 2016-12-21 by alexrj