slic3r/Slic3r #3282: Pressure advance overrides print speed

Sebastianv650, 2016-03-20:
First, a big thanks for all the hard work to alexrj and all the other developers!
I was playing with the pressure advance setting in the last days, and once calibrated properly it's working realy good. I'm using 0.1 for a gregs wade extruder now.

I'm using the latest Master code with this config:
[config.txt](https://github.com/alexrj/Slic3r/files/181380/config.txt)
Test cube:
[20x20x10.stl.txt](https://github.com/alexrj/Slic3r/files/181383/20x20x10.stl.txt)

But I noticed a bug regarding the print speed, it looks like the printer is sometime printing slower than expected when pressure advance is enabled. I had a short look into the gcode, you can see the error here:

G1 Z0.540 F9000.000 ; move to next layer (2)
G1 E12.52740 F2100.000 ; pressure discharge
G92 E12.53477 ; restore E
G10 ; retract
G92 E0 ; reset extrusion distance
G1 X158.323 Y146.823 F9000.000 ; move to first perimeter point
G11 ; unretract
G92 E0 ; reset extrusion distance
**G1 F2700**
G1 E0.03596 **F2100.000** ; pressure advance
G92 E0.00000 ; restore E
G1 X139.677 Y146.823 E0.24848 ; perimeter
G1 X139.677 Y128.177 E0.49695 ; perimeter
G1 X158.323 Y128.177 E0.74543 ; perimeter
G1 X158.323 Y146.763 E0.99311 ; perimeter
G1 X158.760 Y147.260 F9000.000 ; move to first perimeter point
**G1 F1350**
G1 E0.96614 F2100.000 ; pressure advance
G92 E0.99311 ; restore E
G1 X139.240 Y147.260 E1.25324 ; perimeter
G1 X139.240 Y127.740 E1.51336 ; perimeter
G1 X158.760 Y127.740 E1.77349 ; perimeter
G1 X158.760 Y147.200 E2.03282 ; perimeter
G1 X158.374 Y147.156 F9000.000 ; move inwards before travel

--------------------

I think this way the feed of the pressure control overrides the real intended feed rate?
PS: Please excuse some maybe wrong wording, I give my very best but english is not my first language ;)

markwal, 2016-03-20:
When you build from master, it's helpful if you also include a commit number (first 8 characters) because "master" will be meaningless as time goes on and looking up the commit later based on your report time is a PITA.

Or you can use "git describe --tag" from the command line.  It'll give the nearest tag followed by a sufficient amount of the current commit identifier to show where you're at.

Sebastianv650, 2016-03-20:
My problem is I'm not using git, I downloaded the zip package and compiled it. Looking at the commits, it should be 5ff7511 if this is the right number. Sorry, it isn't the latest version as stated above..

I found out that my example with the test cube above isn't quite good. This one is better, because it shows why it is not every time only printing at retract speed:

Model: [mq-belly.stl.txt](https://github.com/alexrj/Slic3r/files/181403/mq-belly.stl.txt)
Config: [config_belly.ini.txt](https://github.com/alexrj/Slic3r/files/181404/config_belly.ini.txt)

This is a extract of a single layer (happens at every one, just a random one to show you):
[SingleLayer.txt](https://github.com/alexrj/Slic3r/files/181405/SingleLayer.txt)

As you can see, it is doing a pressure advance at line 8 and the discharge at line 21. The next two perimeter blocks from line 29 to 39 don't have a pressure advance so they are printed at the correct speed, but there should be a pressure advance.
At line 46 it is doing a pressure advance for the outer shell of the object, but at line 78 where the perimeters are finished the discharge block is missing! Instead, it is doing another advance at line 81 for the infill which is again printed at the retract speed. But only until line 103, where the speed is reset due to a travel move.

So we have two problems in reality:
.) Speed used for pressure advance overrides real print speed.
.) In some cases, pressure advance isn't activated. Maybe this happens when the print speed since the last advance event wasn't changed, but this is wrong as pressure was discharged before.

closed 2017-03-05 by alexrj