slic3r/Slic3r #3283: Setting for Retraction speed greyed out with FW retraction and Pressure advance enabled

Sebastianv650, 2016-03-20:
Another one regarding Pressure control:
Slic3r version: Master branch

When FW retraction is used, the field for retraction speed is greyed out. If Pressure control is enabled (not 0), it uses the retraction speed for it's pressure regulation moves. In this case, the field should be enabled I think.

whosawhatsis, 2016-03-20:
The field probably shouldn't be greyed-out, because the user might want to set retraction speed in the start code like this:

M207 F[retract_speed]

You might say it doesn't make much sense to use firmware retraction in that case, but that's the way I do a lot of things until the printer has a way to store those values. It still gives me the ability to alter those variables without reslicing, and to use the firmware feature of doing the un-retract at a different speed (which works really nicely, and might be worth considering for Slic3r, too).

alexrj, 2016-03-21:
@Sebastianv650, good catch.

@whosawhatsis, I see your point but both retract_speed and start_gcode are defined within the same Printer Settings preset, so it's pointless to use a placeholder. If you need to supply a value to `M207` you can write it directly in the start_gcode. Disabling fields is a key part of making the interface more friendly for the average user, since it provides immediate feedback about what effect other options have. Since any option might be used as a placeholder, we might want to leave everything always enabled...

whosawhatsis, 2016-03-21:
@alexrj A better way to do that is with the warning icons we discussed in another thread. There is so much variation in machines and configurations that disabling fields is risky (as the OP's issue demonstrates), but a warning icon attached to a dialog box will avoid this type of conflict while doing a better job of providing the feedback you mentioned.

alexrj, 2016-03-21:
@whosawhatsis, my point is that options should be disabled when Slic3r does not use them, which is not an ambiguous criteria. The OP case is just a bug (when Pressure Advance is enabled retract_speed *is* used). In other cases there's nothing to warn about :)

whosawhatsis, 2016-03-21:
Even if you do disable the input, I would still put one of these icons next to any disabled input. Maybe not a warning symbol, but maybe a question mark or info symbol. Clicking on the icon should pop up an explanation of why the box has been disabled, and which setting(s) need to be changed to re-enable it.

alexrj, 2016-11-22:
@Sebastianv650, I fixed this. Thank you for reporting.

closed 2016-11-22 by alexrj