slic3r/Slic3r #3285: Support "raft" layer too sparse

OskarLinde, 2016-03-21:
### Version
1.2.9

### Operating system type + version
OSX 10.11.3 (15D21)

### Behavior
Using "pillar" support
[image]

The central support piece of this layer is printed in mid air.

[image]

Previous "raft"-like layer is missing  material underneath:

[image]

lordofhyphens, 2016-03-21:
This can be verified with pretty much any model (inverted pyramid probably would show the first layer).

What looks like to me is that it's using the raft settings foe the first layer instead of support. On small pillars it becomes pretty likely that the support is, well, unsupported. This leads to failures :(

EliamSylin, 2016-03-24:
I have the same issue here.

Version 1.2.9
Windows 10 Home

[image]

results in 

[image]

If I turn on one layer of raft. it tightens everything up.

[image]

and it looks like this.

[image]

I've tried this for all the different support patterns with the same result. 

It's as though slic3r is ignoring the interface pattern spacing, unless raft is enabled.