slic3r/Slic3r #3286: Memory Leak?

Octribin, 2016-03-22:
### Version
1.2.9

### Operating system type + version
Xubuntu 15.10 64 bit

### Behavior
[image]

I can see excessive memory usage when modifying options or changing the printer setting forcing a recalculation of the print, especially as there is a complex or large print. It's annoying as I want to place two copies of a model (see attachment) that are supposed to be printed quite accurately. Of course I expect higher memory usage than normal, but in the situation of an input stl file sized around 360 KiB and an output gcode file sized below 20 MiB I can't imagine it needs more than 1 GiB of RAM to slice. I think it's noteworthy that the option "Delete All" does not free any RAM. It would be useful to store previous states of the planned print to undo some steps, but as far as I know Slic3r does not offer the option to undo, so saving the history of the session would not make much sense.

Reproduction:
* use my settings (see below)
* load the .stl file (see below) to Slic3r
* wait for Slic3r to finish the calculations
* optionally rotate the object by 90° around z-axis and wait for Slic3r to finish calculations

#### STL/Config (.ZIP) where problem occurs
[Memory leak.zip](https://github.com/alexrj/Slic3r/files/183294/Memory.leak.zip)

alexrj, 2016-03-26:
1.2.10-dev (stable branch in git) contains a few good fixes to memory leaks.

Octribin, 2016-03-31:
It looks like the "Wipe while retracting" printer setting for my first extruder causes the issue together with the nozzle diameter greater than 0.4 mm as I tested it out...

alexrj, 2017-03-13:
Master branch (1.3.0-dev) handles memory even better. I'm closing this as we do a constant optimization work, it's implicit.

closed 2017-03-13 by alexrj