slic3r/Slic3r #3290: Avoid crossing perimeters crosses between layers

buckaroob8, 2016-03-24:
### Version
1.2.9

### Operating system
Windows 10 64 bit

### Behavior
The "avoid crossing perimeters" works great within a layer, but it disregards borders when going to the starting location for the next layer.  I've attached the STL, generated GCODE, INI settings for a project that produces these results, and a screenshot of the tool path showing the crosses over the borders.  

By the way, these are internal borders.  Is this the intended behavior, i.e. is the "avoid crossing" function only supposed to avoid crossing external borders?

#### STL/Config (.ZIP) where problem occurs
[butterfly.zip](https://github.com/alexrj/Slic3r/files/187238/butterfly.zip)

mrvn, 2016-06-07:
I think for the next layer the z value has already changed and slic3r assumes the nozzle is then clear of all obstacles. Is you problem caused by overhangs or corners curling up and the head then hitting them?

buckaroob8, 2016-06-07:
My main problem is that I have a head that oozes a lot so it's dragging strings across the open area.  If it could stay within the border between layers, it would produce much nicer prints.

lordofhyphens, 2017-01-11:
As best as I recall, "Avoid Crossing Perimeters" should avoid travelling across any loops that are of the "Perimeter" type. Obviously if there's no way to do so then it will anyway.

In the general case it's a hard routing/pathing problem. You could probably include a call to the same code used to generate the travel moves (passing the previous layer). 

https://github.com/alexrj/Slic3r/blob/d4d26a5dbbb43f0a8a0cccd176eb8c9bf431b58a/xs/src/libslic3r/GCode.cpp is a reference.

If you wanted to avoid *all* perimeters all the way down it's a much more restrictive problem formulation and will generate some weird paths, IMHO.

cobra18t, 2017-02-20:
I know this is an old thread, but should the "Avoid Crossing Perimeters" avoid crossing small perimeters? I have a number of small ~3mm holes that the nozzle runs straight through rather than going around as I would expect with this option checked.

lordofhyphens, 2017-02-20:
It should be avoiding them (unless it doesn't have a choice). Unless you're
referring to crossing perimeters from the previous Z layer, that really
doesn't belong in this issue (and should be in a new issue with
reproduction steps).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

cobra18t, 2017-02-20:
Thanks, I will open a new issue with more specific details since it is crossing internal small perimeters on its current layer.

cobra18t, 2017-02-23:
The hole run through issue with avoid crossing perimeters checked can be found here:
https://github.com/alexrj/Slic3r/issues/3721

rufogil, 2018-03-12:
Hi. Yes, it is definitely crossing perimeters, most of the time when changing to a new layer, but also inside the layer. From time to time I used to increase the nozzle temperature to improve the adhesion between layers depending if I need a stronger part (like mechanical parts) and this causes the nozzle to produce fine strings across borders. I'm using Windows version
[filterCover_STL_and_G.zip](https://github.com/alexrj/Slic3r/files/1804691/filterCover_STL_and_G.zip)
[filterCover_STL_and_G.zip](https://github.com/alexrj/Slic3r/files/1804694/filterCover_STL_and_G.zip)

 1.3.0-dev. The model I'm printing is a really good example of the need to avoid crossing internal perimeters. It is a part of a special smoke filter using s 120mm fan.