slic3r/Slic3r #3291: make contact z distance for support configurable

jfpion, 2016-03-26:
slic3r 1.2.9
win 7 64

the contact z distance is none or 0.2

with some plastic and some configuration the distance is not enough for the support to easy pull out

but there is no way to configure it

it could be a good option to be able to increase it as needed

thank you

Paul--W, 2016-03-27:
Note that the Contact Z control is a combobox.
You can enter a value.  For instance, I use .07mm there.

lordofhyphens, 2016-03-27:
This really needs to be in the manual too  alexrj/Slic3r-Manual#33 should go at least part of the way to "fixing" this. Maybe the UI element needs to be altered to be something else instead (a text-box with prefilled 0.2 maybe)? Could add 0 to the tooltip then.

jfpion, 2016-03-27:
no it is not adjustable in my version (win7 64 1.2.9) you can change it but the change is not kept:
go to support tab , change the contact z distance to 0.4 , click on an another box to be sure the value is 0.4 then click on an another tab like infill for exemple, then back to support. the value is back to 0.2 

that's why i request to make it adjustable

or is there something I missed ...

jfpion, 2016-03-27:
there is something i missed, but i think there is a slight pb in the UI. to keep the value you want you need to save it just after the change before adjusting an another tab, it is the only one with this behaviour, and that's why i though it wasn't adjustable.

lordofhyphens, 2016-03-27:
I think changing the UI element to a pre-filled text box instead of a dropdown would handle this better IMHO. This is a common enough "issue" that I think it warrants the change.