slic3r/Slic3r #3293: [Proposal] More efficient GCode export

cangelis, 2016-03-26:
Current GCode export method sometimes takes forever when there are many islands in the input. Please correct me if I am wrong, this is the pseudo code of current flow of perimeter generation:

```
// perimeter generation
// part of process() of PerimeterGenerator
foreach $island in $region {
    // perimeter collection is the collection that holds all the perimeters of the layer
    $perimeterCollection->append(generate_perimeter($island));
}
// gcode export
// part of process_layer() in Slic3r/Print/GCode.pm
// group extrusions by extruder and then by island
foreach $perimeter in $perimeterCollection {
    // loop to find out the island of the perimeter
    foreach $island in $layer {
        if ($perimeter is in $island) {
             // group $perimeter
        }
    }
}
```

If we say perimeter count = p, and island count = i
Current algorithm is `O(p * i)`

There is no problem if there are only a couple of islands but if there are hundreds or thousands of islands, the export may never end.

For example this model: http://www.thingiverse.com/thing:399474 has about 1000 islands in each layer so there are 1000 perimeters in each layer. The algorithm needs to iterate 1000*1000 times to group the perimeters (same for infills).

What I propose is to push each perimeter to the collection in its own island. So only one iteration would be enough. Pseudo code:

```
// perimeter generation
// part of process() of PerimeterGenerator
foreach $island in $region {
    $island->perimeterCollection->push(generate_perimeters($island));
}
// gcode export
// part of process_layer() in Slic3r/Print/GCode.pm
// group extrusions by extruder and then by island
foreach $island in $layer {
    // group $island->perimeters
}
```

So the algorithm does not need to iterate to find island of the perimeters again because perimeters are already in the island object as a collection.

If we say island count = i

Current algorithm is `O(i)` rather than `O(i*p)`

I really would like to create a pull request rather than just explaining it but I am not good at Perl and XS, anybody knows Perl, XS and codebase can implement this in an hour I think.

alexrj, 2016-03-27:
Thank you for taking your time to contribute with this idea. It makes sense, of course. I remember looking already into this, and then giving up. I think there's a reason behind the current behavior, but I don't remember it off the top of my head and I'd need to check the code. Will do.

cangelis, 2016-03-27:
Thank you. I have been researching about the performance issues of gcode export and this is my conclusion. I also try to implement my proposal but looks like it will take some time. 

Please let us know if you make any progress.

lordofhyphens, 2016-05-29:
@cangelis We'd be glad to review a PR if you get something working. I'm going to move it into the "futures" milestone for now, if @alexrj decides to work on it he can pull it out of that milestone. :)

bubnikv, 2016-05-29:
I think I could look into this issue. From my experience and from the experience of my college, this seems to be quite a bottleneck.

cangelis, 2016-05-30:
Ah sorry, currently I don't have any milestone. @bubnikv please let us know if you make any progress.

bubnikv, 2016-05-31:
I looked into the code shortly. A simple search structure over the bounding boxes of polygons like an octree or kd-tree would help a lot, but there may be a better solution assigning an island ID to each perimeter / infill somewhere earlier in the process. I will try to come up with something.
Vojtech

ledvinap, 2016-05-31:
@bubnikv : You should try profiling current code first. 1M iterations at 1000 instructions each will take less than 1 second ..

bubnikv, 2016-05-31:
Peter,
it is not the number of iterations, but the additional "inside polygon" test, which makes it slow IMHO.
I agree though, that profiling is a good thing. Indeed, I have a history of speeding up a proprietary 3D mesh Boolean library multiple times by doing the profile/improve cycles over and over for months.
Vojtech

bubnikv, 2016-06-01:
Gang,

I ran the code through a profiler and I have found a subtle bug causing superfluous copying of C++ data to perl in an inner loop. Namely, just asking for a number of slices by calling $#{$layer->slices} causes a copy of the C++ object. Factoring the copying out of the inner loop made the slicing time of the referenced thingiverse object http://www.thingiverse.com/thing:399474 to drop from 28 minutes to 3 minutes and 7 seconds on my high end Windows laptop. The main changes are in Print/GCode.pm. The changes are straightforward. Apart from factoring some stuff out of the inner loop, I also added a pre-filtering by a bounding box. The bounding box test is applicable to many other spots in Slic3r code.

I don't know how to create a pull request, if one of my pull requests is already in the queue. The changes are available in the github/prusa3d/Slic3r branch under a checkin bb9c3dd9cea918da8478b606bdc38b1b34b4ea75

When slicing this particular model with  --no-only-retract-when-crossing-perimeters --no-extra-perimeters, the slicing time drops from 3 minutes to 1 minute (from the original 28 minutes). Indeed, both these functions have terrible time complexity on objects with a plenty of islands. We at prusa3d do not use these two features. We don't need to worry about crossing the perimeters, because the E3D extruder leaks very little and the extra perimeters feature seems to be flawed in case of models with more than a single island. From the code it seems that one compares a total length of all conturs in the one layer up against a length of a single contour (of a single island). That does not make sense to me, but I may have understood the code incorrectly.

Vojtech

lordofhyphens, 2016-06-01:
I will cherry pick the changes.

@bubnikv what I usually do is start at master and make a new branch for a
fix or feature, then start making changes, committing to that branch. If I
needed to bring those changes into another branch, well, that is what
merges are for :)

Otherwise git cherry pick is how you tease out individual commits from one
branch to another.

cangelis, 2016-06-06:
@bubnikv Thank you very much. Your patch works great.

lordofhyphens, 2016-06-30:
It's merged.

closed 2016-06-30 by lordofhyphens

alexrj, 2016-07-10:
@bubnikv, thank you for the patch and @lordofhyphens, thank you for crafting it.