slic3r/Slic3r #3295: Gapfill causing large negative feedrates in output

lordofhyphens, 2016-03-27:
### Version
Slic3r 1.3.0-dev 9576973b578f0c39f8cbd73c225187ba65b6cef4

### Operating system type + version
Debian Linux 8 x64

### Behavior
Output gcode contains large negative feedrates before all gapfill moves. 

_Expected Results_
No feedrate should <=0, because it's impossible.

 _Actual Results_
*Feedrates like G1 F-199232 or G1 F-3.58987e+06

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
[negfeedrate.zip](https://github.com/alexrj/Slic3r/files/191047/negfeedrate.zip)

alexrj, 2016-12-21:
I just tested this with current master and issue looks gone. If you have a chance to test yourself we can close this issue.

lordofhyphens, 2016-12-22:
Well my test case no longer shows the issue.

closed 2016-12-22 by lordofhyphens