slic3r/Slic3r #3296: 1.2.9 Windows 10 x64 - Fan control layer timing is way off.

drifter57, 2016-03-29:
### Version 1.2.9
### Windows 10 Version 1511 (OS Build 10586.164)
### Behavior
The fan controls do not appear to function as I expect.  I set the controls to not start the fan unless layer timing falls below 40 seconds per layer, yet I am seeing the fan turn on for layers that take well over 2 minutes to print.  Am I not understanding how this feature is supposed to work, or is there actually a problem with the layer timing in the program?

This has occurred in every STL where I have enabled the cooling feature.

markwal, 2016-03-30:
One thing to keep in mind is that Slic3r has to compute the layer time based on commanded speeds.  Your bot will not actually go commanded speeds due to firmware limits and acceleration.  Usually the overhead of these things is more on the order of 50% increase or so, so a commanded 40 second layer taking 1 minute would be pretty typical (depending on length of extrusions, number of fine details, etc.).  But if you're commanding speeds above the speed your firmware will go, the discrepancy will get greater and greater.

However, in your case, it may just be that something else about your settings is overriding this.  For example, keep fan always on, bridges fan speed, etc. will all override.  While it has occurred in every STL where you've enabled it, it's still helpful to include settings and STL for people who might try to reproduce the behavior you're seeing rather than making them try to reverse engineer what you're doing.

drifter57, 2016-04-19:
If the software is not capable of accurately executing the command, why include the control?  I don't think I'm setting speeds above my firmware capabilities, but then I'm not deeply familiar with the math involved in stepper calculations either.  I'm using the stock Marlin on a TAZ4 with a Rambo board.  Supposedly the TAZ4 can run at 200mm/sec.  My max is set at 180mm/sec travel.  I have never had it skip steps.  Not sure if that info helps or not.

Is there a way to test my system and set some type of constant so that the timing selected in the configuration is accurate, or at worst close?  A 50-100% deviation from the setting seems unacceptable in my mind.  Is this an adjustment feature that needs to be added to Slic3r?

I have attached my standard Slic3r configurations, and two .stl files.  The 2slider.stl file should show the fan timing problem.  Another fan issue is represented in the bridgetest.stl file.  This was printed using Repetier and Slic3r 1.2.9.  Watching the parameters on Repetier as the gap is bridged at layer 19 the fan speed drops to 52% when it is set for bridging at 100%.  This does not adequately harden the bridged filament.  

A strobe verifies that the fan speed drops.  I have set max speed at 85%, which I'm thinking is max for normal printing, and push it to 100% for bridging.  If the command does not function in this fashion please let me know.

[FanProblem.zip](https://github.com/alexrj/Slic3r/files/225619/FanProblem.zip)

lordofhyphens, 2016-04-19:
This may have been related to #3134. Could you try to reproduce your symptoms with the version in git? You will need to follow the instructions for building from source. 

The other thing I can think of is to check the output gcode for M106 commands. If it is outputting the expected commands then it isn't a problem with Slic3r.

drifter57, 2016-04-19:
Hmmm... For the second problem at least, Gcode looks OK.  There are quite a few fan speed changes in quick succession in this section of the code.  It appears the hardware and Repetier just can't respond quickly enough and I'm seeing an average speed.  Would it be better perhaps to have the fan speed consistent for the entire initial bridge layer instead of switching up and down over the various, sometimes very short, lengths of filament?  I guess it would be necessary to assess the effect to adhesion on the "grounded" portion of the bridge layer.

lordofhyphens, 2016-04-19:
You could write a small post-processing script to reduce/remove fast
changes in the fan speed. And from what I have seen in my own experience,
some fans just don't like being PWM'd for fine speed control.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-05-29:
The speed changes have been merged in, and I think that the build on prusa3d.com has that code in it, could you give it a try (if you can't build from source)?

I'm going to close this for now, reopen if you find it is still a problem.

closed 2016-05-29 by lordofhyphens