slic3r/Slic3r #3299: [Proposal] Layer change inside part

pieniacy, 2016-04-07:
Is it possible to add "Infill" among "Seam position" options?
I did a little search on this and it seems there is no slicer implementing such thing, is it for some reason difficult to do?

lordofhyphens, 2016-04-07:
1) You're not guaranteed to have infill.
2) Perimeters are normally printed before infill, which IIRC is the cause
of the "seam", which is a relatively slow Z move up to the next layer from
the previous layer.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

pieniacy, 2016-04-07:
1) ok, but most of the time we print with infill - hollow or very small parts happen rarely
2) when i imagine best seam-hiding path i see something like:
* starting layer inside print at calculated seam point
* printing part of infill
* printing perimeters
* finishing infill and ending roughly where seam point is
* going up at that point

mrvn, 2016-04-11:
I thought seam was where each perimeter starts and stops. Nothing to do with when it moves one layer up.

If its actually only where the Z change happens then that setting seems rather pointless. Half the printers will have lift so moving the head anywhere will change Z anyway, a lot more than one layer. From the other half 99% you have infill and the Z move up would be over infill and again the setting would be pointless. And for the rest it would far more sense to move away from the outer perimeter for the Z move if that is what you call seam.

pieniacy, 2016-04-11:
Ok, i get your point. Thanks :)

I have an idea of continuously printed perimeter, but i will work on it a
bit and see how it works.

lordofhyphens, 2016-04-11:
That is vase mode :)

pieniacy, 2016-04-11:
no, no, not vase, i mean normal layer by layer printing, but setting paths
so that while being on perimeter no stops are made

lordofhyphens, 2016-04-12:
Generally each perimeter is printed completely on its own if possible, as a loop, so I'm not quite sure what you are asking for now. :(

Drmaestro, 2016-05-03:
I think what he meant was creating a continous loop which includes the perimeters and the infills. If there is a certain amount of overlap between the infill and perimeter, the printing head can transition from the infill to the perimeter without any interruption. It can start at the infill, and then go to the perimeter and come back to the infill again for beginning of the next layer. I think Cura does something similar if set at the options (infill before perimeters) and this could be helpful for external layer quality. It is true that not all of prints have infills, but it could be an option for prints which have an infill. This might be more complicated than it sounds however, as it might require changing the slicing logic.

lordofhyphens, 2016-05-03:
@Drmaestro You can already tell Slic3r to print infill before perimeters
(and external perimeters before internal perimeters).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

thinkyhead, 2016-05-06:
As an alternative, even (or especially) when printing perimeters, the last few mm of one layer should do a "smooth transition" to the next layer instead of stopping XY and moving Z straight up. This would be essentially the same as the "Joris" or "vase" style of printing, but just for the transitions. The slicer could account for the ramping up to the next layer with a gradually-increasing E flow, and a corresponding ramping-down of E flow when finishing that first perimeter.

This could potentially eliminate the "Z scar" altogether, smearing it out over more space.

pieniacy, 2016-05-08:
@Drmaestro exactly! sorry, maybe i couldnt express it properly
@thinkyhead i like your idea as much as mine, it also should solve the problem, but i imagine it would be much much more difficult to implement

thinkyhead, 2016-05-09:
To me, implementing a slicer seems like magical voodoo.
So I bet @alexrj would have a lot of "fun" figuring it out.

mrvn, 2016-05-09:
Slic3r generates closed loops for the perimeters. But then it has to break open that loop to actually print it. Where it breaks it the head will move from one perimeter to the next or from perimeter to infill or to the next layer. Doesn't really matter where it goes, point is that it breaks the loop often leaving a scar.

It might be beneficial for slic3r to never (unless impossible to avoid) lift/retract at the end of the outer perimeter. Instead the head should move inwards and then lift/retract  That way any scar caused by the lift/retract would be inside where it isn't seen.

Similar with un-lift and un-retract. Don't do that on an outer perimeter. Do that somewhat inside and then start the perimeter loop by printing towards the outside edge. A single perimeter would then look like this:
__...
|
X
|__...

gege2b, 2016-05-09:
Hi
Good idea, but I read another suggestion similar to this one a few years ago, and if my memory is good, @alexrj said that this way of handling perimeters start/stop is patented 

https://github.com/alexrj/Slic3r/issues/1037#issuecomment-18401135

lordofhyphens, 2016-05-18:
http://www.fabbaloo.com/blog/2013/5/14/stratasys-solves-those-troublesome-3d-print-seams.html

http://www.google.com/patents/US20130095303 

AFAIK it's patented, so it can't go into Slic3r.

lordofhyphens, 2016-05-18:
Closing this issue. Feel free to reopen if you find that the patent gets invalidated.

closed 2016-05-18 by lordofhyphens

a4jp-com, 2016-05-19:
A z-lift with wipe pretty much gets rid of seams so you wouldn't need the internal bump that is shown in the patent.

mrvn, 2016-05-24:
Also the patent is for additive methods. I want this for my CNC mill for subtractive construction. :) Those seems on the inside of holes are ugly.

Patola, 2017-02-20:
@lordofhyphens prusa slic3r just implemented "rear" as an option to seam position, which is exactly what seam hiding / the stratasys patent deals with. Should it be backported to slic3r or will it be forever absent?
[image]

KISSlicer also implements it as "destring".
@bubnikv @alexrj 
[image]

bubnikv, 2017-02-20:
> prusa slic3r just implemented "rear" as an option to seam position, which
is exactly what seam hiding / the stratasys patent deals with. Should it be
backported to slic3r or will it be forever absent?

The "rear" seam is something else. It means, that Slic3r will position the
seam close to max Y while still avoiding overhangs and trying to put the
seam in the corners.

Patola, 2017-02-20:
@bubnikv thanks for the clarification and sorry for the confusion.

lordofhyphens, 2017-02-20:
Depends on how involved @bubnikv made the implementation ;)

I can definitely see a use for some better control of seams, although the
described behavior sounds like it might have some surprises lurking  (this
sounds mostly like aligned, except biased towards max Y).

bubnikv, 2017-02-20:
> I can definitely see a use for some better control of seams, although the
described behavior sounds like it might have some surprises lurking (this
sounds mostly like aligned, except biased towards max Y).

Yes, it is exactly that, aligned towards the max Y. The issue with the
implementation is, it calculates a linear weighted sum of various cost
functions, therefore it may not do what you would like it to do. This is a
perfect task for a neural network to learn the intents. Some volunteer to
take the challenge? :-)

lordofhyphens, 2017-02-20:
Yeah, I could think of a few corner cases where it would surprising. Probably still worth including :), assuming I can port it over.