slic3r/Slic3r #3300: bottom solid infill speed ?

jfpion, 2016-04-07:
slicer 1.2.9

i think i found a bug
it is about the top solid infill speed, the default speed is 15 which is slow, but for a nice finish not a pb.

but is there a reason the same speed is used for the bottom infill (ie the x first layers -depends of the bottom solid layer ? as it is flat against the bed the finish is perfect no need to go so slow (except the first one , but that an another story)

or may be it is done for the raft user ?

anyway it is not a real pb just a surprise as it is said top solid infill and not top/bottom solid infil 

thank you

lordofhyphens, 2016-04-07:
Bottom solid infill is all flat undersides as I understand it, not just the
part against the build platform.

jfpion, 2016-04-08:
ok so that's not a but juste a misleading name of the option