slic3r/Slic3r #3301: 403 Forbidden

usernamevalis, 2016-04-09:
Hi 

When I try download Slic3r I get a 403 Forbidden error - the url is :http://dl.slic3r.org/mac/

Thanks

lordofhyphens, 2016-04-09:
Not seeing the problem when I just  tried; probably just a transient issue with the host or something. Try again :)

closed 2016-04-09 by lordofhyphens

dchosking, 2016-09-26:
I get this same issue on the windows links, I have not been able to download it for months.
Other users are having this same issue
https://www.reddit.com/r/3Dprinting/comments/4pof4w/can_anyone_upload_the_latest_slic3r_for_me_the/

lordofhyphens, 2016-09-26:
Don't know what is going on, as it just worked for me.

lordofhyphens, 2016-09-26:
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev

For 1.3.0-dev  (as it comes off of the build server).

dchosking, 2016-09-26:
[image]

This is a screen shot of what i get when trying to download, seems like an authentication issue to me

lordofhyphens, 2016-09-26:
I literally just did a download with no authentication. On my phone. It's something transient at best. At worst it's not something I can do anything about. 

[image]

dchosking, 2016-09-26:
It is strange, i just tried on my phone and it seems to work just fine. 
My phone is even running on the same network as the pc but i cant seem to download it.

lordofhyphens, 2016-09-26:
I'd look at your PC/browser, I think we can rule out the server itself in this case.

dchosking, 2016-09-26:
Right, i am going to grab my work laptop and give that a go, it was formatted around a week ago so should be a good test

lordofhyphens, 2016-09-26:
I don't know what to tell you, I can't reproduce the problem and I have no access to debug the server itself. Your options are to get the OSX build from Prusa3d.com or build 1.3.0-dev from git or keep digging.

https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X

drakruka, 2017-01-19:
I had The same Problem , turned out They were Blocking my country's IP , So i used A VPN and it worked!

rspencer01, 2017-08-06:
This seems to be a persistent problem.  Recognizing that you don't have access to the server, but that this is a pain for people from certain areas (I had to ssh into a server I owned in a different country), perhaps it might do to put a disclaimer near the link at the least?  Or a link to a mirror?  I'd also perhaps suggest to your host that blocking certain IPs is exceptionally bad practice, as it is hurting your (and hence their) brand.

lordofhyphens, 2017-08-06:
@alexrj

<https://github.com/alexrj/Slic3r/issues/3301#issuecomment-320523259>, or mute
the thread