slic3r/Slic3r #3304: Ubuntu 16.04 does not have libwxgtk2.8

AndKe, 2016-04-12:
..but libwxgtk3.0  - however, it won't run:

```
andre@loke:~/RepRap/Slic3r129/bin$ ./slic3r --gui
Can't load '/home/andre/RepRap/Slic3r129/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so' for module Wx: libwx_gtk2u_adv-2.9.so.3: cannot open shared object file: No such file or directory at <content>/DynaLoader.pm line 157.
 at /home/andre/RepRap/Slic3r129/lib/std/Slic3r/GUI/AboutDialog.pm line 6
Compilation failed in require at /home/andre/RepRap/Slic3r129/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
BEGIN failed--compilation aborted at /home/andre/RepRap/Slic3r129/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
Compilation failed in require at /home/andre/RepRap/Slic3r129/lib/std/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at /home/andre/RepRap/Slic3r129/lib/std/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 71) line 1.

```

AndKe, 2016-04-17:
apparently something to do with the way I backed+up/restored on bigger SSD, some links were omitted.
Anyway - it works fine with libwxgtk3.0

closed 2016-04-17 by AndKe

cliffgi, 2016-07-26:
I am having this problem also.  I'm using Linus Mint 18 which has libwxgk3.0, not 2.8.  Slic3r fails to install, 'can not find libwxgtk2.8', or words to that effect.  AndKe above simply states "works fine with libwxgtk3.0" I don't understand how, if you can't get slic3r1.2.9 to install in the first place.  libxgtk2.8 will not install on ubuntu16/LinuxMint18, so that is not a possible solution.   If anyone out there and spell out the steps to install Slic3r1.2.9 on ubuntu16/LinuxMint18 (libxgtk3.0), I sure would be appreciative.  CG

AndKe, 2016-07-26:
I did not install it, just uncompressed to a folder..

lordofhyphens, 2016-07-26:
@cliffgi Mint should have 1.2.9 in your apt repository. 

Otherwise see the wiki for building instructions, which has been updated recently. I suggest using the Ubuntu 16.04-specific instructions.

cliffgi, 2016-07-27:
Thanks.  I'll try to hunt down the compiled...  Only saw the 
build/compile method.

CG

cliffgi, 2016-07-27:
Holy expletive deleted!  It just showed up in Mint's Software Manager.  
SM is always my first choice.  I have no idea what the package name is 
for slic3r via apt  install.   Unfortunately, just tried SM install and 
it put a nice icon out there that does nothin. I'll run down the install 
log tomorrow to see if I can wrench it into submission.  I suspect 
trusty libxgtk2.8, aaaagain.

CG

lordofhyphens, 2016-07-27:
Package name should be "slic3r". Additionally there is always apt-cache
search to look up package names.

cliffgi, 2016-07-27:
Super thanks.

cliffgi, 2016-07-28:
Still no luck getting slic3r 1.2.9 to run on Linux Mint 18.  It is in 
the official Mint repos and am able to install using both Software 
Center and Synaptic Package Manager (as well as apt-get) but in all 
cases application will not start.  WIth SC and SPM, the icon created 
does nothing but pulsate a couple of times.  apt-get installs some png 
files in /usr/share, but does not appear to have an executable.  Under 
no circumstance is are there any files created in my user directory, 
hidden or otherwise.  I think a .slic3r is normally supposed to be 
created.? It might be worthy to note that I fob-booted LinuxMint 18 
LIVE, installed slic3r via SC, and got the very same app no-launch 
problem.   Having to reboot to Win7 for slic3r gives me the 
drizzling-runs.  HATE WINDOWS!

Knowing where to find the install log files that might help me work 
through this.  Any thoughts.

thanks Cliff

lordofhyphens, 2016-07-28:
Any "install log files" that would have been created is the provision of
your distro. I can't help there. There is something going on with your
package (please report to the maintainer).

If you are up to the adventure, suggest looking up the install instructions
in the wiki for running from git head (which, IMHO, is superior to 1.2.9).
Be sure to use the local::lib approach so you don't trash your local Perl
modules.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

cliffgi, 2016-07-30:
Thanks.  I would think that the Mint 18 LIVE bootable would successfully run slic3r if installed right from it's included distro and built-in Software Manager, but it does not.  This eliminates anything I may have dorked on my install, since the Live is a completely independent and pristine boot.  Does not give me a warm-fuzzy that anyone with Mint 18 (based on Ubuntu 16) can run slic3r.  Only thing is that maybe Mint18's LIVE Software Manager does not fully install all prerequisites.  If this be the case and this be the real world... time to fire the Software Manager.  I think I'm going to create a VM Mint18 and start fresh.  If it works, I'll rebuild my system, AGAIN!  MTF

cliffgi, 2016-08-05:
Thanks Alex.

I have discovered that I can successfully run old versions of compiled 
slic3 ...tar.gz up to version ...64_1.2.6(experimental) and 
x86-1.2.7(experimental).  Anything newer fails to launch.  The problem 
may also exist for users running Ubuntu 16, upon which Mint 18 is 
based.  It is definitely slic3r problem as slic3r 1.2.9 won't run even 
from Mint 18 Live-Boot.

It is a shame that there does not appear to be a method for feedback 
contact to development team for this great product.  I'd love to donate, 
but as I can not communicate feedback, am not confident about a fix 
anytime soon for Mint18(Ubuntu 16).

Until then, I'll have to be content to use x86_64-1.2.6(experimental).

Thanks for all your time.
Cliff

lordofhyphens, 2016-08-05:
I'm not Alex. Alex is @alexrj.

If you're on Linux, then you can build from git source. We helpfully provide step-by-step instructions for that.

If the ubuntu/mint 1.2.9 package is broken, then contact the package maintainer for that package and tell them! We don't maintain the packages for each individual distro out there. The maintainer's information is listed with the package description (check Software Manager) and file a bug with them. 

As for the packaged version from slic3r.org, I'd need to know what the error messages you're getting to even try to figure out what is going on.

> It is a shame that there does not appear to be a method for feedback
> contact to development team for this great product.  I'd love to donate,
> but as I can not communicate feedback, am not confident about a fix
> anytime soon for Mint18(Ubuntu 16).

What are the numerous responses I have made, white noise? 

This is even to what seems to be an unrelated problem (as the OP closed his own issue several months ago), to what appears to be distro-specific. 

Please write up a fresh, clean bug report against the slic3r.org package (NOT your distro package). You can refer back to this issue (#3304).

lordofhyphens, 2016-08-05:
From README.md:
> You can also find me (rarely) in #reprap and in #slic3r on FreeNode with the nickname Sound. Another contributor, LoH, is also in both channels.

lordofhyphens, 2016-08-05:
Ubuntu package page: https://launchpad.net/ubuntu/+source/slic3r