slic3r/Slic3r #3306: [feature request] Fan control for support material

Zuborg, 2016-04-17:
A support material often adhere to model very hard. And sometimes it can be stronger than printed model. It may be very useful if we may to cool support structures when printing. I suggest add fields for the fan control section: 1. fan speed for support structure; 2. fan speed for model perimeter parts printed over interface layers; 3. fan change mode lead time in milliseconds (this option is useful because of mechanical inertia of a fan rotor). Every option can be switched off with negative value. If it active, it alter current fan mode when printing support structures or model's perimeter which placed above support.

lordofhyphens, 2017-03-27:
@Zuborg did you ever do any experimental work with this? If you turn on verbose gcode and come up with a test object that shows the problem (when printed) and how your proposal would mitigate that problem... I could understand it better and it would have a better chance of getting done.

Zuborg, 2017-04-03:
This issue is very close to the old [#1539](https://github.com/alexrj/Slic3r/issues/1539). When I print any model with ABS and with highest temperature (for increasing a model strength), the support and model fused in one monolith body. But if I trying to print some object with full power of the cooling fan (centrifugal fan with a narrow nozzle) that does it fragile and its layers could be delaminated easy. That does impossible using a steady full power for complete print of the model with support.  I've didn't special experiments but I think that's a good idea to use a temporary cooling enforce for improve a weakness of support structures.