slic3r/Slic3r #3307: Feature Request: enable support modifier

mrvn, 2016-04-18:
I have an object where at one side I need support because I have a 90° overhang that won't print any other way. But at the same time on the other side I have holes that must not be blocked or even marked by support and that print fine without.

It would be ideal if I could load a modifier volume and only turn support on for that volume.

lordofhyphens, 2016-04-18:
Duplicate of #2570

hoegge, 2018-02-07:
So is this the active feature request for this? Would be great if you could turn support on / off, change "from build plate" only and all the other support (overhang e.g. ) parameters with modifiers and generics. That would really ease work flow a lot - and of course add the feature to Slic3r Prusa Edition too. Thanks

lordofhyphens, 2018-02-08:
@hoegge no that is #2570

Also we don't control what goes into the prusa3d fork of Slic3r

closed 2018-02-08 by lordofhyphens

hoegge, 2018-02-08:
Why did you close the feature request?

hoegge, 2018-02-08:
Ah seems you reopened the other request instead. I know you don't work on PE but Alex does, right. Wondering - does it really make sense with two different Slic3r forks. Maybe one (I) should consider pushing PE features back into Slic3r

lordofhyphens, 2018-02-08:
@hoegge Alex doesn't work on Slic3rPE either.

The Prusa3d fork got made because Alex had disappeared and they had wanted to push stuff out. Kinda hard to do that when the person with the only commit access is MIA. A month or so later I came on but they had already done so much in different places that it pull requests aren't enough. You have to do a port and undo some of the things we don't want while keeping the things we do.

It is not trivial to pull their work, just FYI.

Some things do get pulled in, but they have people whose job it is to work on Slic3r (and Prusa isn't paying to port it upstream in the form of pull requests) and we here who are volunteers. I haven't been nearly as active as I would want lately because of my own day job being at a crunch time.

lordofhyphens, 2018-02-08:
@hoegge also #2570 has never been closed itself (I just checked), so not sure what you were talking about here (lots of issues that were linked to it were closed though).

lordofhyphens, 2018-02-08:
Further discussion about this feature should go to #2570.