slic3r/Slic3r #3308: Fan Kickstart - Feature Request

mirage335, 2016-04-18:
At low speeds, fans may be able to continue running, but unable to start moving. Please add a feature to briefly kick the fan at full power, before operating at low speed.

Prepending M106 commands with a configurable G4 (dwell) command might do the trick.

TTN-, 2016-04-20:
This is something that ought to be implemented in the firmware. The slicer should be able to ask for a fan to start, the firmware can worry about how to do that.

Currently marlin supports it already. Have a look in the advanced configuration settings file.

lordofhyphens, 2016-05-17:
It's also fixable with a post-process script.

lordofhyphens, 2016-05-28:
The post-process script would look for M106 S[1-9]+[0-9]* commands and insert a new line M106 S255 right before it.

lordofhyphens, 2016-05-28:
Here's a python script to do just that.

#!/usr/bin/python
import sys
import re

sea =  re.compile("M106 S[1-9]+[0-9]*")
rep = re.compile("M106 S255\n\g<0>")
out = open(sys.argv[1]+"_fixed", 'w');
with open(sys.argv[1]) as f:
    for r in f:
        if re.search(sea, r) is not None:
            out.write(re.sub(sea,"M106 S255\n\g<0>",r))
        else:
            out.write(r)

closed 2016-05-28 by lordofhyphens

mirage335, 2016-05-28:
Thanks for the solution. Not quite what I was looking for, but quite helpful nonetheless. Thanks again.

lordofhyphens, 2016-06-02:
Fixed via #3353

lordofhyphens, 2016-06-02:
@mirage335 well, it's just the post-process script I posted, so it's in the main repository now :)

mirage335, 2016-06-02:
Indeed, that helps. Personally, prefer such functionality exists upstream instead of on my personal machine.