slic3r/Slic3r #3309: Filament Specific G-Code - Feature Request

mirage335, 2016-04-18:
Please allow Start/End/BeforeLayer/AfterLayer/ToolChange custom G-Code scripts to be appended under filament settings. Particularly for RepRapPro firmware, it may be desirable to manually set standby temperatures, allow cold extrusion, etc.

Regarding standby temperature configuration, this is also a useful workaround when the shells generated by ooze prevention settings are undesired.

lordofhyphens, 2016-05-27:
I'll see what I can do; it'll probably end up being an cut'n'paste from the Printer Settings gcode. Hope it doesn't get abused :D

lordofhyphens, 2016-05-28:
Done, going to close for now. It's not merged in yet because there's no test for it yet (and I suck at writing test cases).

mirage335, 2016-05-29:
Thanks, looking forward to using this!

lordofhyphens, 2016-05-29:
Sorry, confused it with something else. Reopening sp I don't forget about it.

As for the order,  what sounds most useful? For Start/end, I was thinking about having it look like this

P-start f-start (printing code) f-end P-end

For before/after layer change, I think I am going to use the same scheme. For everything else the filament code will precede ther printer code.

mirage335, 2016-05-29:
Probably best just to let filament specific gcode proceed (after) the generic printer code. If something like temperature has been set to a default in the generic printer code, then a specific type of filament might need to override this every time the tool switches back to a tool loaded with special-temperature filament.

lordofhyphens, 2016-05-29:
Yeah that's what I was thinking of, the positioning is to maintain the
order around the gcode in terms of how general.

VanessaE, 2016-09-28:
I was thinking today that this could be exploited to allow for per-filament autotemp as well.  That would require the per-filament "start" to happen as late as possible, like right before the first layer change (so that it happens after the M109 command Slic3r adds).

lordofhyphens, 2016-09-29:
Except that if you extended it to the check of "hey we need to add m109" check it wouldn't matter for that

bubnikv, 2016-10-27:
@lordofhyphens Would you please point me to the check-in with your filament specific G-codes, so I could review it and possibly merge with the Prusa3D fork? Thanks

lordofhyphens, 2016-11-26:
@bubnikv PR is up. 

@mirage335 there's the pull request, a win32 build should show up at https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches in an hour or two.

lordofhyphens, 2017-03-11:
PR was merged, closing.

closed 2017-03-11 by lordofhyphens