slic3r/Slic3r #3310: Feature Request - Generate support only attached to bed.

mozgy, 2016-04-18:
Slic3r 1.2.9

please add option to generate support material only attached to bed.
Example would be pipes (or tubes) in + combination. Support should be generated only from bed to outer surface and not any support generated inside pipes.

lordofhyphens, 2016-04-18:
Pretty sure this is a duplicated request. Will check later when I am not on
my phone. I know that completing this would involve a rewrite of the entire
support algorithm system.

You should probably start thinking of approaches and their pros/cons.

jfpion, 2016-05-05:
yes is an old request. seems it could be approached with the discuted horizontally space adjustment for the support. if you specify a gap large enough it will be no support in the small hole/tube

Patola, 2016-05-05:
If that requires a rewrite, that would be a very good opportunity to try
and make supports in zig-zag so that they can be easily detachable by
pulling it from one side, like in simplify3d and cura!

lordofhyphens, 2016-05-17:
@Patola the support material patterns are all implemented in Perl, take a look at how it's done and you can probably work out how to make a zig-zag. :)
I will happily review any PRs that come in for that.

Similar to how we already have pillars, rectilinear, etc.

Patola, 2016-05-18:
Hey, I would love it indeed. However, I can't understand the code as it is
now and I find it difficult to find a gentle introduction to
slicing-related algorithms. Do you have any study material to suggest?
Please try and be specific, as I already know that generically there's lots
of path-finding and graph algorithms.

lordofhyphens, 2016-05-18:
Start by studying what's already there; how the existing routines generate
the toolpaths your are looking at (afaik all of the support is generated in
Perl).

lib/Slic3r/Print/SupportMaterial.pm is a good place to start looking. You
don't even have to recompile to see changes because it's in Perl. Just get
it to build from source once and you can tweak the Perl files and reload to
see changes.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-08-30:
Odds are the place to start is going to be from the Pillars support structure, as the capital construction is more or less what we're looking for here.

I haven't had time to dig through and understand what the current algorithm expects w/r/t data structures in (perl's weak typing makes this a bit opaque, which is very irritating).

lordofhyphens, 2016-10-04:
@bubnikv provided an implementation for this.

Source: https://github.com/lordofhyphens/Slic3r/tree/infill-over-plate
The build server should have it ready in a bit, and I'll get a PR up after I've looked it over.

lordofhyphens, 2016-10-04:
@mozgy Try out the new PR/branch (should be up on bintray for win32).

PR #3525 (2017-03-26): A new feature "support_material_buildplate_only" implemented.
Also a bug has been fixed for zero interface layers. Before
slic3r would put infinite number of interface layers over top surfaces,
if the number of interface layers was set to zero.

Fixes #3310 
Fixes #2757
Thanks to @bubnikv for this work.

Win32 automatic development build: https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches/1.3.0-infill-over-plate-192#files
closed 2017-03-26 by alexrj