slic3r/Slic3r #3311: Enabling gap fill greatly increases print time w/ autospeed

jhnphm, 2016-04-21:
### Version
510ca9f9e2f0e550ee3733510f7738e6ca0128aa 

### Operating system type + version
Arch Linux

### Behavior
Autospeed enabled (speeds set to 0) and nonzero gap fill greatly slows down print speed compared to autospeed w/ gapfill = 0

#### STL/Config (.ZIP) where problem occurs
[3DBenchyTest.zip](https://github.com/alexrj/Slic3r/files/229265/3DBenchyTest.zip)

#### GCode diff screenshot ####
[image]

lordofhyphens, 2016-04-21:
While I'm not that surprised as the autospeed logic is trying to maintain pressure, could you try it with current -dev? There was some changes w/r/t cooling and the timers involved.

jhnphm, 2016-04-21:
I was running 510ca9f which already seems to include that and is the latest from master that I can tell. Adjusting the gap fill speed does not seem to improve times (tried to slow it down to 2 in order to decrease required pressure during gap fill and allow the rest to speed up didn't work, and neither does increasing the speed.

lordofhyphens, 2016-05-17:
Gap fill speed = 0 disables gapfill, so yeah it'd be an interaction between the logic to maintain pressure.

lordofhyphens, 2016-06-30:
Okay, near as I can tell it's calculating the minimum mm^3 for the different extrusion entities and those calculations look like they're including gapfill. The referenced pull request removes gapfill amounts from calculation in all cases. 

Please test and see if this helps you.
https://github.com/lordofhyphens/slic3r/tree/fix-3311-no-gapfill-autospeed-effects

lordofhyphens, 2016-06-30:
Yeah, it's fixed, at least with my quick test with your config and settings.
With the new PR the time is now 2hrs as guesstimated by Pronterface, down from 10hrs.

alexrj, 2016-07-10:
Wait, what's the point in using autospeed if pressure is changed all the time because of gap fill?

lordofhyphens, 2016-07-10:
Since pressure would be lost anyway on a layer change, if gap fill was delayed until the last print moves then it would not matter.

Additionally the ridiculously slow speeds causes heat damage to the print in practice.

hyperair, 2016-09-04:
Perhaps there should be a minimum speed tweakable instead? The issue here is that some gapfill moves and some thin wall moves have really puny widths. Trying to satisfy "max print speed" and "all moves should have the same E speed" can result in printing at 5mm/s or slower for the entire print.

simonkuehling, 2017-04-12:
@alexrj technically speaking your question is absolutely valid in general. But I guess we should take a second to define what the main goal of autospeed/ constant pressure was in the first place. From my understanding there are two main reasons why autospeed is a very clever idea:
  - *quality improvements* - avoiding pressure changes between infill/perimeters/etc successfully removes all the surface artefacts introduced by rapidly switching between different flow rates.
  - *maximizing print speed/ high output printing* - in many cases where maximizing the deposition rate of the print is the main goal (large layer thickness, huge nozzles), it is very convenient to just use a combination of autospeed feedrates + cooling thresholds as the only variable parameters while tuning a print job for minimum print time.

For both of these cases, it would be totally ok to calculate the maximum volumetric speed without taking gap fill movements into account. As long as the gap fill extrusions will also be limited to the max flowrate, there are usually no bad side effects from accepting a pressure change here. Especially if these are the last print moves of a layer (like mentioned above). The only case in which this may have effects on surface quality is thin walls with varying thicknesses where the gap is filling directly between single perimeter lines. But on the other hand, it is much more favorable to get a properly filled wall instead of deactivating gap fill all together and beeing left with a void on the inside that leads to structural weakness of the part.
Slic3r's Gap Fill logic is absolutely outstanding, powerful and unseen in all the other slicing engines out there!

If I think about a real world scenario that would interfere with a variable flow rate introduced by gap-fill moves, there is only one that we might consider:

  - an extrusion system that is unable to react to quick pressure changes with its extrusion system at all. Paste extrusion with compressed air feeding for example.

It seems like a huge tradeoff for this case as well, though. For the ratio of gap fill moves compared to the rest of the perimeters/infill/support, it should be a good idea to come up with a hardware solution instead of slowing down all prints by a factor of 10, I guess...

What do you think? Maybe I am missing something here?

lordofhyphens, 2018-04-03:
@Timor take a look at the pull request by @hyperair