slic3r/Slic3r #3313: Wrong thin infill, too much extrusion

Tinchus2009, 2016-04-23:
### Version
_Version of Slic3r used goes here_
1.3.0-dev (1.2.9-243-g510ca9f) but also the same happens in 1.2.9 stable

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
1.2.9-243-g510ca9f

### Operating system type + version
_What OS are you using, and state any version #s_

Win 10 64bits

### Behavior
I sliced this model and started printing. The printing failes 5 times, always about the same time, something around when model reached 5 mm height.
The sixth tie I started to print, I stayed to watch carefully. When the printing reached the critic part, I sould see a lot of extrusion being made on a small area. This of course made a mountain of melted plastic and in the next layers the noozzle just crashed against this pastic.
Looking at the sliced simulation, I can see this small area, filled with thin lines. But the problem is the all my extrusions are manually set to be 0.69 (perimeters, external perim., infill, etcs, everything should be 0.68). So what it looks it is happening is that this space is being filled with like 20 thin lines, but with an extrusion of 0.68 wich is toooo much for such small area. Here is a picture. Atched .stl file and config
[config_slicer.zip](https://github.com/alexrj/Slic3r/files/233197/config_slicer.zip)

[image]
[Spool_spindle.zip](https://github.com/alexrj/Slic3r/files/233193/Spool_spindle.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Mikk36, 2016-04-24:
This looks like bridging. In the config you provided, the extrusion widths were set to be auto.
The width of those traces is calculated to be the same as nozzle width, so check that over.
If you must, you can decrease bridge flow ratio.

Tinchus2009, 2016-04-24:
Ok... if that is bridging (it should not because there are like 4 other layers beneath), and that bridge is calculated to be 0,4 (my noozle), why is extrudeing like hell? I mean, the extusion was like meant to be 0,68.

Mikk36, 2016-04-24:
Are you sure there are 4 layers beneath it? From the image it looks like bridge with nothing below it.
You also conveniently left out the layer height part from the image (right side).

Tinchus2009, 2016-04-24:
What do you mean with conviniently????? Sorry, I croped the image to redice sice. 
Also, ok, we can argue if bridges belong there or not (i think not), but lets asume they are bridges: 
1.- there is nothing set to auto, al extrusions are set to be 0.68, I have re atached the config bundle, also I have atached a screen shot showing the extrusion configuration, everything is set to 0.68. lso, bridges flow rate is 1
2- if they are bridges, why are they being calculated to be 0.4, but what I see in real life is something being extruded like meant to be 0.68. The amount of plastic deposited there is HUGE.
 I tached some more pictures of 3 consecutive layers. 

[Slic3r_config_bundle.zip](https://github.com/alexrj/Slic3r/files/233656/Slic3r_config_bundle.zip)
[image]
[image]
[image]

Tinchus2009, 2016-05-02:
Hi. I have checked with ther files. And yes, what is seen in the pictures is bridging, but the problem then is that all extrusion was meant to be 0.68 , and this bridgin is been made thninner. IS that the correct behaviour?

Mikk36, 2016-05-02:
Bridging overrides any extrusion widths you may have set because an extrusion width can only be reached when you squeese a new layer on top of a previous one.
In case of bridging there is no previous layer below it so it goes into a bridging mode, where the plastic will come out of the nozzle at the same width as the hole in the nozzle.

Tinchus2009, 2016-05-02:
oh.... you are so right!!!! thanks

closed 2016-05-02 by Tinchus2009