slic3r/Slic3r #3314: Feature request: dual extrusion anti-ooze walls

VanessaE, 2016-04-29:
### Version
git from commit 510ca9f9e2f0e550ee3733510f7738e6ca0128aa (with LoH's Debian tweaks)

### Description
Currently, Slic3r has dual extruder ooze reduction by way of adjusting inactive nozzle temperature and retraction thereof.  These help, but in practice, it doesn't work well enough.  This Youtube video demonstrates a method that appears to get rid of the last of the ooze:  https://www.youtube.com/watch?v=10zlqHJavcg

The g-code presented therein does not work properly under Slic3r, at least with my otherwise well-calibrated Prusa i3.  It's easy enough to add "sacrificial walls" to the plate and configure Slic3r to wipe the head across them after every tool change, but that doesn't allow me to configure when the walls are drawn, relative to the rest of the print (from what I understand, they should be the first thing printed, followed immediately by a wipe action if a tool change has to be executed before moving over to the rest of the print).

So, my proposal is to add a feature to Slic3r to properly generate the "sacrificial wall" seen in the video (but at a more appropriate location and orientation) and perform the wipe action done to it.

It's hard for me to explain it any better than this, so maybe just watch the video. :smile:

As for an object to test such a feature with, grab this Thing, http://www.thingiverse.com/thing:124450 or perhaps something more complex.  The description section of the Youtube video URL I gave references this Thing: http://www.thingiverse.com/thing:643780

VanessaE, 2016-05-01:
Follow-up:  I've been playing around with this idea.  I can't write code that'll go into Slic3r, but I came up with a model file (AMF), some "after layer change" g-code, and "tool change" g-code that seem to work ("before" g-code is empty).  They depend on having a suitably-large tool-change retract.

```gcode
; "after layer change" code goes here
G91					; relative positioning
G1 Z-[layer_height] E-3
G90					; absolute positioning
G1 X70  Y6 F15000;
G4 S2
G1 X130 Y8 F1000;
G1 X130 Y18 F1000;
G1 X70  Y16 F1000;
G1 X130 Y14 F1000;
G1 X70  Y12 F1000;
G1 X130 Y10 F1000;
G1 X70  Y8 F1000;
G91
G1 Z[layer_height] E3
G90
; end of "after layer change" code

; "tool change" code goes here
M104 S160
G91					; relative positioning
G1 Z-[layer_height]
G90					; absolute positioning
G1 X70  Y6 F15000;
T[next_extruder]
M109 S[temperature]
T[previous_extruder]
M109 S185
T[next_extruder]
G1 X130 Y8 F1000;
G1 X130 Y18 F1000;
G1 X70  Y16 F1000;
G1 X130 Y14 F1000;
G1 X70  Y12 F1000;
G1 X130 Y10 F1000;
G1 X70  Y8 F1000;
G91
G1 Z[layer_height]
G90
; end of "tool change" code
```

See below for my complete config and the model file I'm testing with:

[Prusa i3 - Slic3r 1.3.0 config bundle.ini.zip](https://github.com/alexrj/Slic3r/files/244377/Prusa.i3.-.Slic3r.1.3.0.config.bundle.ini.zip)
[sacrificial-walls-dual-extruder.amf.zip](https://github.com/alexrj/Slic3r/files/244378/sacrificial-walls-dual-extruder.amf.zip)

lordofhyphens, 2016-05-17:
You could write a post-processing script; those have access to most of Slic3r's variables.

VanessaE, 2016-06-01:
This option isn't needed - turning on ooze control in "Print Settings" enables a sacrificial skirt that slic3r wipes the nozzles on, in addition to temperature variation.

closed 2016-06-01 by VanessaE