slic3r/Slic3r #3315: spacing between infill/infill and perimeter/perimeter #3315

richardbarkotz, 2016-04-29:
### Version
1.2.9 

### Operating system type + version
Windows 7 32 bits

### Question
Hello,

I´m trying to understand how is the math for calculating spaces between the extruded filaments. I hope someone can help me.
In the website http://manual.slic3r.org/advanced/flow-math I found the following formula:
spacing = extrusion_width - layer_height * (1 - PI/4)

I verified through the G-code if this is indeed the spacing and found out that this only applies to the spaces between contours to contours. 

The space between infill to infill (using rectiliear fill patern and 100% infill) is different. Here I´m actually only analizing the top solid layer (since the real fill pattern for other layers is honey comb)

Hier it is some examples:
the wall is 5mm thick 
2 contours were used
tip size: 0.4mm

**1 CASE**
road width    layer height   calculated spacing       real space contour/contour      real  space infill/infill
    0.6                    0.1                    0.579                               0.579                                     0.695
real space infill/perimeter
          0.578

**2 CASE**
road width    layer height   calculated spacing      real space contour/contour        real space infill/infill
    0.6                    0.4                    0.514                               0.514                                     0.482
real space infill/perimeter
          0.482

1- So what happens hier is that in the 1 CASE the spaces between infill/infill are larger than the road width, so there won´t be any overlapping, causing gaps on the top layer infill (which should be solid).
2- Sometimes "space perimeter to infill" takes the value "space contour/contour" (1 CASE) and sometimes it takes the value "space infill/infill" (2 CASE)

Can someone help me out how is the "space infill/infill" calculated?? 
when is the "space infill/perimeter" equals  the "space infill/infill" and when is it equals the "space contour/contour" ??

Thanks

VanessaE, 2016-04-30:
I can confirm that solid infill is printing with very slight gaps between adjacent lines.  At first, I thought it was my hardware or filament, but Pronterface's gcode preview shows the same "slightly-porous" effect in solid infill as I'm getting in the actual print.

As for a test .stl, any random cube ought to be enough to see the effect.

lordofhyphens, 2016-04-30:
Pronterface is wrong, as it does not show true line widths in its preview.

richardbarkotz, 2016-04-30:
I calculated the spacing from the gcode, looking were the middle of the lines are positionated...the width from the cases that i showed were all 0,6 (the width is achieved by the quantity of filament extruded in the same point)

richardbarkotz, 2016-04-30:
* looking where

richardbarkotz, 2016-04-30:
calculated spacing= calculated by the formula from slic3r3d
real space contour/contour=calculated from gcode
real space infill/infill= calculated from gcode
real space infill/perimeter= calculated from gcode

alexrj, 2016-05-02:
@richardbarkotz, have you read the source code? Look at Flow.cpp.

closed 2016-05-02 by alexrj